CauldronDevelopmentLLC/CAMotics #113: errors running a new install of current debian linux. the deb file wont work so i tried to build. followed the instructions main page and got the following errors when trying to build.

r01walker, 2015-01-01:


r01walker, 2015-01-01:
i am running a fresh install of debian linux. i tried to install the deb file and it failed so i tried to build as per instructions this site. i got this error. cut and pasted from the terminal. 
user5@user5:~$ git clone https://github.com/CauldronDevelopmentLLC/OpenSCAM.git
Cloning into 'OpenSCAM'...
remote: Counting objects: 1528, done.
remote: Total 1528 (delta 0), reused 0 (delta 0)
Receiving objects: 100% (1528/1528), 8.38 MiB | 143 KiB/s, done.
Resolving deltas: 100% (762/762), done.
user5@user5:~$ cd OpenSCAM
user5@user5:~/OpenSCAM$ scons
scons: Reading SConscript files ...
EnvironmentError: No tool named 'config': not a Zip file:
  File "/home/user5/OpenSCAM/SConstruct", line 13:
    env.Tool('config', toolpath = [cbang])
  File "/usr/lib/scons/SCons/Environment.py", line 1782:
    tool = SCons.Tool.Tool(tool, toolpath, **kw)
  File "/usr/lib/scons/SCons/Tool/__init__.py", line 94:
    module = self._tool_module()
  File "/usr/lib/scons/SCons/Tool/__init__.py", line 154:
    raise SCons.Errors.EnvironmentError(m)
user5@user5:~/OpenSCAM$ scons package
scons: Reading SConscript files ...
EnvironmentError: No tool named 'config': not a Zip file:
  File "/home/user5/OpenSCAM/SConstruct", line 13:
    env.Tool('config', toolpath = [cbang])
  File "/usr/lib/scons/SCons/Environment.py", line 1782:
    tool = SCons.Tool.Tool(tool, toolpath, **kw)
  File "/usr/lib/scons/SCons/Tool/__init__.py", line 94:
    module = self._tool_module()
  File "/usr/lib/scons/SCons/Tool/__init__.py", line 154:
    raise SCons.Errors.EnvironmentError(m)
user5@user5:~/OpenSCAM$ scons strict=0
scons: Reading SConscript files ...
EnvironmentError: No tool named 'config': not a Zip file:
  File "/home/user5/OpenSCAM/SConstruct", line 13:
    env.Tool('config', toolpath = [cbang])
  File "/usr/lib/scons/SCons/Environment.py", line 1782:
    tool = SCons.Tool.Tool(tool, toolpath, **kw)
  File "/usr/lib/scons/SCons/Tool/__init__.py", line 94:
    module = self._tool_module()
  File "/usr/lib/scons/SCons/Tool/__init__.py", line 154:
    raise SCons.Errors.EnvironmentError(m)

jcoffland, 2015-01-02:
You missed one very important part of the build instructions.  Look again at the part about building C! and setting ```CBANG_HOME``` to the location of the C! build.

closed 2015-01-02 by jcoffland