CauldronDevelopmentLLC/CAMotics #114: Tool selection not working

digmanwaves, 2015-01-02:
The files below are supposed to mill three slots in a workpiece, using three different tools.
 - [testsetup.xml](https://github.com/digmanwaves/misc/blob/master/OpenSCAM/Tool-selection-not-working/testsetup.xml)
 - [testsetup.ngc](https://github.com/digmanwaves/misc/blob/master/OpenSCAM/Tool-selection-not-working/testsetup.ngc)

However, the tool changes seem not to take effect.
I hope you can reproduce the effect. Thanks for your time.

bye,

Walter.

I'm running a version of openscam, described by the console output below:
10:48:17:INFO(1):************************** OpenSCAM **************************
10:48:17:INFO(1):     Website: http://openscam.org/
10:48:17:INFO(1):   Copyright: (c) 2011-2014
10:48:17:INFO(1):      Author: Joseph Coffland <joseph@openscam.org>
10:48:17:INFO(1):Organization: Cauldron Development LLC
10:48:17:INFO(1):        Args: testsetup.xml
10:48:17:INFO(1):      Config: <none>
10:48:17:INFO(1):*************************** Build ****************************
10:48:17:INFO(1):     Version: 0.2.5
10:48:17:INFO(1):        Date: May 7 2014
10:48:17:INFO(1):        Time: 11:01:28
10:48:17:INFO(1):  Repository: Git
10:48:17:INFO(1):    Revision: 16899f3c6cf9505328dcd897505077e6732fdc06
10:48:17:INFO(1):      Branch: master
10:48:17:INFO(1):    Compiler: GNU 4.8.2
10:48:17:INFO(1):     Options: -std=gnu++98 -O3 -funroll-loops -mfpmath=sse -ffast-math
10:48:17:INFO(1):              -fno-unsafe-math-optimizations -msse2
10:48:17:INFO(1):    Platform: linux2 3.12-1-amd64
10:48:17:INFO(1):        Bits: 64
10:48:17:INFO(1):        Mode: Release
10:48:17:INFO(1):*************************** System ***************************
10:48:17:INFO(1):         CPU: Intel(R) Core(TM) i5-2520M CPU @ 2.50GHz
10:48:17:INFO(1):      CPU ID: GenuineIntel Family 6 Model 42 Stepping 7
10:48:17:INFO(1):        CPUs: 4
10:48:17:INFO(1):      Memory: 7.43GiB
10:48:17:INFO(1): Free Memory: 3.55GiB
10:48:17:INFO(1):     Threads: POSIX_THREADS
10:48:17:INFO(1):  OS Version: 3.13
10:48:17:INFO(1): Has Battery: true
10:48:17:INFO(1):  On Battery: false
10:48:17:INFO(1):  UTC Offset: 1
10:48:17:INFO(1):         PID: 13417
10:48:17:INFO(1):         CWD: /home/wdaems/lab/Cyclopes-Arges/Camera/testsetup
10:48:17:INFO(1):**************************************************************
10:48:17:WARNING:././testsetup.ngc:9: Not implemented: G91.1 (IJK Incremental Distance Mode)
10:48:17:INFO(1):Tool Path Bounds: ((0,0,10), (80,40,30))
10:48:17:INFO(1):Tool Path Dimensions: (80,40,20)
10:48:17:INFO(1):AABBTree boxes=12 leaves=12 height=5
10:48:17:INFO(1):Computing surface bounded by ((-1.42275,-1.42275,-1.42275), (101.423,51.4227,19.4227)) at 0.711379 grid resolution in 4 boxes
10:48:17:INFO(1):Time: 0.30 secs Triangles: 70728 Triangles/sec: 235039.21 Resolution:

jcoffland, 2015-01-02:
The problem is that parenthesis denote comments in G-Code.  So where you've written:

    (T2)
    (M6)

You should have written:

    T2
    M6

Otherwise, those lines are interpreted as comments and have no effect.

closed 2015-01-02 by jcoffland

digmanwaves, 2015-01-02:
Damn... sorry to have wasted your time on this. 
Stupid of me that I did not see this stupid fault of mine. 
I even can't blame it on my lisp-history... I rarely program lisp anymore. 

Thanks, 

Walter.

jcoffland, 2015-01-02:
No problem.