CauldronDevelopmentLLC/CAMotics #115: Crashes if a file used in a project is moved

whitequark, 2015-01-05:
With the following log:
```
22:45:16:ERROR:Exception: Accessing './Downloads/1st_side(engraving).tap': No such file or directory
22:45:16:ERROR:Exception: SmartPointer: Can't dereference a NULL pointer!
```

jcoffland, 2015-01-06:
I'm having trouble reproducing this.  What version if OpenSCAM and what operating system are you using?  Also, what were the circumstances?  Can you give me a step-by-step way of reproducing this bug?

whitequark, 2015-01-06:
OpenSCAM 0.2.5 on Debian wheezy.

To reproduce:
1. add any g-code file to the project
2. close OpenSCAM
3. delete the file
4. run OpenSCAM as `openscam foo.xml`, where `foo.xml` is the project file

jcoffland, 2015-01-06:
This seems to be fixed in the latest code.  I'm working on a new release.

Thanks for the report.

closed 2015-01-06 by jcoffland