CauldronDevelopmentLLC/CAMotics #116: G28 return to zero uses incorrect relative Z?

cmason1978, 2015-01-10:
Hi.  Thanks for awesome OpenSCAM!

I'm trying to understand the rapid move I've indicated with the gray arrow here:

[image]

It makes it look like the tool will rapid very close to the part, which is not what I want.  I'm trying to figure out if this is a real issue with the G-code or with OpenSCAM's simulation.  I'm leaning towards the latter, but my interpretation could easily be wrong.

It seems to be generated by this G-code where we transition from one toolpath (an adaptive clearing operation that ends at the right hand side of this edge), and move back to zero before starting the new toolpath (a contour operation beginning in the lower right hand corner of the view):
```
...
X2.0173 Y1.6457 Z-0.3625
G0 Z0.5                      ( Move to retract height. )
M5                           ( Spindle stop. )
G28 G91 Z0                   ( Return to machine zero through Z0 in relative mode. )
G90                          ( Absolute mode. )

(Contour1)
M9                           ( Coolant off. )
T2 M6                        ( Tool 2.  Automatic tool change. )
S16542 M3                    ( Spindle speed 16,542 rpm.  Spindle on clockwise. )
M9                           ( Coolant off. )
G0 X2.5256 Y-0.1876          ( Rapid. )
Z0.5                         ( Move to retract height. )
Z-0.3638                     ( Plunge. Shouldn't this have plunge feedrate? )
G1 Y-0.1873 Z-0.3665 F60.57
...
```
(Comments above are my interpretation of the G-code, not output by post processor.)

Any thoughts here?  I'm reading [this article](https://www.cncci.com/resources/tips/how%20g28%20works.htm) about G28, and [some](http://camforum.autodesk.com/index.php?topic=5949.0) threads on the inventor HSM forum.  My read is that it should remain at Z 0.5 (retract height) because of the G91 in the G28.  I completely admit to not understanding this fully.

I've put the complete program [here](http://www.cmason.com/tmp/wood/curved_joint_bottom_2.nc). It was generated by Fusion 360 (Inventor HSM) with the `generic-grbl` postprocessor.  I'm using OpenSCAM 0.2.5. 

I'm going to attempt to run a shortened version of this on the actual machine (Shapeoko 2, Grbl 0.9g) and see what happens.  It may be hard to tell when air cutting though.

Thanks!

cmason1978, 2015-01-10:
I've tried air cutting [this shortened]() sample on the machine.  It **seems** like it moves **diagonally** from the retract height at the right side of the edge to the machine zero.

Open SCAM seems to simulate this as a straight line at z=0?

That said, I don't think the G-code is right.  I think it should be explicitly moving to the retract height between operations.  I'm going to post to the Inventor HSM forum to see if I can get some help making a post processor that does this.

jcoffland, 2015-01-11:
Believe the problem is that OpenSCAM does not yet implement incremental distance mode.  See issue #42.

Your best option is to make the move explicit as you suggested.

I'm closing this as a duplicate of #42.

BTW, that screenshot looks really cool.

closed 2015-01-11 by jcoffland