CauldronDevelopmentLLC/CAMotics #118: OpenSCAM source install fails Fedora20

MrUmunhum, 2015-02-18:


MrUmunhum, 2015-02-18:
installed v8-devel needed and done.
now get these errors:
g++ -o build/openscam/qt/AboutDialog.o -c -std=gnu++98 -O3 -funroll-loops -mfpmath=sse -ffast-math -fno-unsafe-math-optimizations -msse2 -DNDEBUG -D_REENTRANT -DHAVE_EXPAT -DHAVE_PTHREADS -DHAVE_LIBSQLITE -DHAVE_OPENSSL -DHAVE_CBANG -DUSING_CBANG -DGLEW_STATIC -DQT_CORE_LIB -DQT_GUI_LIB -DQT_OPENGL_LIB -DQT_SHARED -I/src/OpenSCAM/cbang/src -I/src/OpenSCAM/cbang/include -I/usr/include/QtOpenGL -I/usr/include/QtGui -I/usr/include/QtCore -I/usr/include/freetype2 -Isrc -Ibuild build/openscam/qt/AboutDialog.cpp
build/openscam/qt/AboutDialog.cpp: In constructor 'OpenSCAM::AboutDialog::AboutDialog()':
build/openscam/qt/AboutDialog.cpp:34:41: error: invalid use of incomplete type 'class Ui::AboutDialog'
 AboutDialog::AboutDialog() : ui(new Ui::AboutDialog) {
                                         ^
In file included from build/openscam/qt/AboutDialog.cpp:21:0:
build/openscam/qt/AboutDialog.h:29:21: error: forward declaration of 'class Ui::AboutDialog'
 namespace Ui {class AboutDialog;}
                     ^
build/openscam/qt/AboutDialog.cpp:35:5: error: invalid use of incomplete type 'class Ui::AboutDialog'
   ui->setupUi(this);
     ^
In file included from build/openscam/qt/AboutDialog.cpp:21:0:
build/openscam/qt/AboutDialog.h:29:21: error: forward declaration of 'class Ui::AboutDialog'
 namespace Ui {class AboutDialog;}
                     ^
build/openscam/qt/AboutDialog.cpp:39:5: error: invalid use of incomplete type 'class Ui::AboutDialog'
   ui->versionLabel->setText(version.c_str());
     ^
In file included from build/openscam/qt/AboutDialog.cpp:21:0:
build/openscam/qt/AboutDialog.h:29:21: error: forward declaration of 'class Ui::AboutDialog'
 namespace Ui {class AboutDialog;}
                     ^
scons: *** [build/openscam/qt/AboutDialog.o] Error 1
scons: building terminated because of errors.

any ideas?

jcoffland, 2015-02-18:
William, I believe the problem is that your build is using qt-3.3 instead of qt-4 just like in this report for a wireshark build.

Perhaps it will work if you do a clean build and set the PATH like this:

    export PATH=/usr/lib64/qt-4/bin:$PATH
    scons -c
    scons

The key is you need the build system to find the qt 4 tools before the qt-3.3 tools.  Here's some other related information.

jcoffland, 2015-02-18:
MrUmunhum wrote:
> No, that did not work.  Same errors.  Could it be I need a devel file installed?  I only have qt-devel.i686 iunstalled.

jcoffland, 2015-02-18:
Now I see you are running a 32-bit OS.  Try this instead:

    export PATH=/usr/lib32/qt4/bin:$PATH
    scons -c
    scons

If that doesn't work make sure that when you run:

    which uic

It print the path to the qt4 executable not the qt3.3 version.  You can also try running this:

    uic -v

This should print the qt version of the user interface compiler.

You could also try the ```QT_SELECT``` environment variable.  Perhaps this might work:

    export QT_SELECT=4
    scons -c
    scons

MrUmunhum, 2015-02-19:
I am running a x64 Fedora 20 system.

uic -v
Qt User Interface Compiler version 4.8.6
echo $PATH
/usr/lib64/qt4/bin:.:/home/wre00/bin:/usr/lib64/qt-3.3/bin:/usr/lib/qtchooser:/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/usr/sbin:/usr/sbin:/sbin

note: qt4 above not qt-4

still broken.

jcoffland, 2015-02-20:
Previously you mentioned ```qt-devel.i686``` that is a 32-bit package.  If ```uic -v``` is showing 4.8.6 then that is good.  Does it say this if you don't set ```export QT_SELECT=4```?

It's really important that you get a completely clean build with qt4.  Otherwise it wont work.  If you did this:

    export QT_SELECT=4
    scons -c
    scons

And it failed to work, then I'm not sure what's going on.

sorki, 2015-05-11:
Looks like on Fedora 21 you need
```
export PATH=/usr/lib64/qt4/bin:$PATH
export QTDIR=/usr/lib64/qt4
export QTINC=/usr/include/QtGui/
unset QTLIB
scons -c; scons -j8
```
(and correct CBANG_HOME).

Now when I was finally able to build it, I'll try to package it for Fedora.

jcoffland, 2015-05-11:
Ok, that's strange that ```QTDIR``` alone isn't enough.  Thanks for figuring that out.

A Fedora package would be great.  Let me know if I can help.

closed 2015-05-11 by jcoffland