CauldronDevelopmentLLC/CAMotics #119: Awesome - Can I pay a feature bounty?

Analogreality, 2015-03-13:
This is awesome software.. it's really come a long way. Would you be interested in receiving a 'donation' in exchange for implementing new features?

The features would be contributed back into the master branch so that others may benefit.. but I just have a few specific things I would like to add to accelerate my own toolchain.

I plan on donating anyway, but I don't mind helping out more if this is possible.

jcoffland, 2015-03-13:
Absolutely.  What features are you interested in?  Keep in mind though that I've been contracting software development for over 10 years and successfully earn a high per hour rate so I'm not going to jump for a few bucks.  That said, I always deliver what I promise and am eager to help make my software solve people's problems.  So what's on your mind?

Analogreality, 2015-03-14:
I understand your concerns. I'm in the same boat.

After using OpenSCAM for a little while in my toolchain, I think there are a few things that could be done to accelerate productivity.

Here are a few possibilities:
- Additional "/run" or "/play" command line argument to automatically PLAY the simulation, from the beginning, on window load.
- Additional "/fullscreen" command line argument to automatically go into full screen mode, on window load.
- When passed a file via command line, OpenSCAM automatically generates the work area like it normally does, but after completion sets the POSITION to 0, and sets the DIRECTION to FORWARD.
- Background/foreground/workarea color pulled from "style.xml", "style.json", or some other configuration area.

jcoffland, 2015-03-14:
These are pretty easy.  By, "generates the work area," are you referring to the automatically generated uncut workpiece or the final cut workpiece?

Sounds like you want to integrate OpenSCAM in to another app.  If so what's the app?  Is it Open-Source or commercial?

Analogreality, 2015-03-14:
>  By, "generates the work area," are you referring to the automatically generated uncut workpiece or the final cut workpiece?

I mean the 'simulated stock' that gets cut by the simulated endmill. The blue, shiny 3d model that gets deformed.

I'm creating my own development workflow using OpenSCAM.. the idea is that I want it to better integrate into a complete 'design pipe'.

The stuff that I'd connect to it is sort of my own creation, however I am using a few open source things like Inkscape to develop.

In the spirit of open source, however, if we can work something out I would like the requested features to be re-integrated back into this project so the world may benefit from these improvements.

jcoffland, 2015-03-14:
Are you aware that OpenSCAM cannot yet simulate the cutting process step-by-step?  See #2.

Analogreality, 2015-03-14:
Oh, you mean "model of what the stock would look like in realtime at any point in the timeline" vs "model of the final cut piece, with all paths made through it"?

Yes, simulation of move-by-move isn't necessary.

jcoffland, 2015-03-14:
I think the color settings would be best served by a global or per user configuration file.  There are other things, such as a global tool table (See #54) and default units setting, which would be served well by this too.  This is the most difficult part because the correct place to store such a configuration file is different for each OS (Windows, Linux & OSX).  The command line options, ```--fullscreen```, ```--play``` and ```--auto-close``` are all pretty easy.

What kind of bounty are you offering for these features?

Analogreality, 2015-03-14:
I agree, color configuration elsewhere... but just somewhere that can be defined easily. Ideally, web html hex notation.

For these features, how does 40 bucks for ~2 hours of work sound?

jcoffland, 2015-03-14:
Like I'd starve if I had to live off of that.  Sorry, no offense intended.  My going rate is over 7x that but this is a different situation.  Tell you what I'll implement the command line options in the next few days and create a new issue for the color configuration and you can make a donation if you like.

Analogreality, 2015-03-14:
Can you PM me your PayPal address?

jcoffland, 2015-03-14:
You can find donation info here: http://openscam.com/#donations

jcoffland, 2015-03-20:
I've opened #123 for the styling issue.

jcoffland, 2015-03-20:
I implemented ```--fullscreen```, ```--auto-close```, ```--auto-play``` and an extra ```--play-speed```.  I've merged this in to both the *master* and *dev* branches.

There are two other things you mentioned:

 1. #123 
 2. Starting with position at zero and direction forward.

Is the second still necessary when auto play starts at the beginning?  I'm not quite sure what you want this for.

Are you in Linux?  Are you able to build OpenSCAM from source?  I'm not planning on making a new release for awhile.

Analogreality, 2015-03-20:
If --auto-play starts at the beginning and starts play forward in time,
then it is not necessary. That's fine.

I'm using the Windows version myself.  I don't mind building from source,
however I need to learn more about the build tool path for the win32 app
though.

jcoffland, 2015-03-20:
Ok, so #123 is the only remaining issue here.

Re: building on Windows.  The hard part is building the dependencies.  Although I've simplified this recently to some degree.  You still need to build or find pre-built versions of the following:

  * Python
  * scons
  * openssl
  * libv8
  * qt4
  * ciaro

In addition, you need to download the source for boost and set the environment variable ```BOOST_SOURCE``` to the directory where it's unpacked.  All the other dependencies are now included in the [C!](https://github.com/CauldronDevelopmentLLC/cbang) build system.

jcoffland, 2015-09-21:
Fixed in 1.0.0 release.

closed 2015-09-21 by jcoffland