CauldronDevelopmentLLC/CAMotics #125: TPL function to get the current machine state

adamhill1988, 2015-03-22:
There are times when you may wish to get the current machine state so that you can return to that position after doing something.

I don't believe there is any standard gcode for this so it may be necessary to keep track of movements as the program progresses.

jcoffland, 2015-03-22:
In TPL theres an as of yet undocumented function that I added to the ```dev``` branch, ```position()```.  It returns an object like this:

```
  {
    x: <number>,
    y: <number>,
    z: <number>,
    a: <number>,
    b: <number>,
    c: <number>,
    u: <number>,
    v: <number>,
    w: <number>
  }
```

adamhill1988, 2015-03-22:
Ah great, although it might be nice to have an expanded version, something like 'state();' which would include everything, machine things like tool number, spindle speed, coolant etc and software variables like plane, units, incremental/absolute mode etc.

jcoffland, 2015-07-01:
@adamhill1988 all of these state variables are accessible in TPL but via different API functions.  For example, ```feed()```, ```speed()``` & ```tool()```.