CauldronDevelopmentLLC/CAMotics #128: Can,t load some nc files

fredkylam, 2015-03-30:
I can open some but not all nc files, Files created by HeeksCNC are ok but not those by Gcodetools add-on in Inkscape.

jcoffland, 2015-03-30:
Can you provide an example file that exhibits the problem?

GCode is a tricky beast.  There are many versions of it.  Most Open-Source tools use GCode as defined by LinuxCNC.  OpenSCAM [does not implement all the LinuxCNC GCode specification](http://openscam.com/#limitations).

fredkylam, 2015-03-30:
Thank you for the quick response. I attach two files:

cup_of_coffee_0001.nc does not open but umbrella_0001.nc does.

Both are created in Gcodetools under Inkscape. I have no problems running
them in GRBL.

jcoffland, 2015-04-01:
Sorry, I don't see any file attachments.  I think you need to go to the GitHub website and attach files to a comment on this issue.

fredkylam, 2015-04-01:
I am not allowed to attach text files here, but here are my Dropbox links to the two files I mentioned earlier.

https://www.dropbox.com/s/lg5s8lxo3he9onv/cup_of_coffee_0001.nc?dl=0
https://www.dropbox.com/s/siskaciudmn3xtf/umbrella_0001.nc?dl=0

jcoffland, 2015-04-01:
That's odd both files work for me.  I tried with the latest code from GitHub and from the latest binary release (v0.2.5) from the Website.  Both worked as expected.

What exactly do you mean when you say ```cup_of_coffee_0001.nc``` does not open?  Do you see any error output?  Does the program crash?  Can you attach a screenshot?

fredkylam, 2015-04-01:
The file does not even show up in the project box. Maybe I try to reinstall OpenScam and see what happens. Btw I was running under Win7.

jcoffland, 2015-04-01:
Ok, I just tested in Windows 7 with v0.2.5 and it worked there as well.

I know some people have had problems with running OpenSCAM with locales other than US English.

What happens if you run OpenSCAM in a terminal like this?:

```
set LC_ALL=en_US
C:\Program Files\OpenSCAM\openscam.exe
```

Then open the file.

Also see #100.

fredkylam, 2015-04-01:
I have just figured out the problem. Those files which I could not open
reside in a folder with a name which is not  in the Latin alphabet, and
which therefore seems to be not recognized by the program. I moved the
cup_of_coffee...nc file (as well as any others in the same folder) to the
desktop and voila it now works perfectly.

I guess this is a minor quirk I can avoid or live with now that I know the
problem.

Thanks for your time and help in any case.

jcoffland, 2015-04-01:
Thanks for figuring this out.

jcoffland, 2015-10-14:
What do the path names that this fails on look like?  I'd like to resolve this issue but need a test case.  Also, does the problem still occur with the latest version of CAMotics?

fredkylam, 2015-10-15:
I have since renamed the folder and avoided using any names other than in the Latin alphabet. I have forgotten what the original name was, but it was in Chinese, an example is something like this:
..\雕刻機\cup_of_coffee.nc
The three Chinese characters in  name of the foldermean is "CNC machine".
I have not revisited this problem since April so I cannot answer your last question.
I really appreciate your persistence in solving this trivial problem.

closed 2015-10-15 by fredkylam

jcoffland, 2015-10-15:
Thanks for the help.  I think I've fixed this.  It should work correctly in the upcoming 1.0.3 release.

fredkylam, 2015-10-15:
That is great. Thanks for the great work.