CauldronDevelopmentLLC/CAMotics #129: Debian package does not run

karstengit, 2015-04-01:
I installed OpenSCAM as described with 
"gdebi openscam_0.2.5_amd64.deb" 
There where no errors reported, but it does not start.

When i try to run it on a console i see
$ openscam 
openscam: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.15' not found (required by openscam)
openscam: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.14' not found (required by openscam)

I am running
LSB Version:    core-2.0-amd64:core-2.0-noarch:core-3.0-amd64:core-3.0-noarch:core-3.1-amd64:core-3.1-noarch:core-3.2-amd64:core-3.2-noarch:core-4.0-amd64:core-4.0-noarch:core-4.1-amd64:core-4.1-noarch:security-4.0-amd64:security-4.0-noarch:security-4.1-amd64:security-4.1-noarch
Distributor ID: Debian
Description:    Debian GNU/Linux 7.8 (wheezy)

jcoffland, 2015-04-01:
I've got a similar setup with ```libc6:amd64 2.19-13``` installed and it's working fine for me.

What happens if you run this?:

```
sudo apt-get update
sudo apt-get install libc6
```

karstengit, 2015-04-01:
Nothing happens because it is already installed.

ii  libc6:amd64                                       2.13-38+deb7u8                     amd64        Embedded GNU C Library: Shared libraries
ii  libc6:i386                                        2.13-38+deb7u8                     i386         Embedded GNU C Library: Shared libraries
ii  libc6-dev:amd64                                   2.13-38+deb7u8                     amd64        Embedded GNU C Library: Development Libraries and Header Files
ii  libc6-dev-i386                                    2.13-38+deb7u8                     amd64        Embedded GNU C Library: 32-bit development libraries for AMD64
ii  libc6-i386                                        2.13-38+deb7u8                     amd64        Embedded GNU C Library: 32-bit shared libraries for AMD64
rc  libc6-i686:i386                                   2.13-38+deb7u2                     i386         Embedded GNU C Library: Shared libraries [i686 optimized]

jcoffland, 2015-04-01:
Oh, I see the problem.  Your version of glibc is 2.13-38.   This version is too old for OpenSCAM.  You have two options:

  1. Do a dist upgrade: ```sudo apt-get dist-upgrade```
  2. [Install from source](https://github.com/CauldronDevelopmentLLC/OpenSCAM#prerequisites).

karstengit, 2015-04-01:
Yes - this is a problem of Debian stable.
Stable is getting to old - but Testing is not stable.
Up to now i could getting everything to run with stable.

I think i try to compile it from the source.
Hopefully this will go with the "old" system.

jcoffland, 2015-04-01:
The Debian guys are very conservative.  **testing** is quite stable.  Ubuntu and many of the other distros based on Debian use **testing**.  Building from source is not difficult if you follow the instructions precisely.

closed 2015-04-01 by jcoffland

karstengit, 2015-04-01:
The last 2 times i tested the Testing version i could not get X running with an NVidia card.
I want to work with my PC and so i let them more time to stabilize it.

It would be helpful to get the new libc installed in parallel.
But everything is only a symbolic to the same libc names.

karstengit, 2015-04-01:
I tried to do a trick - but it did not work. :-(
I downloaded libc6_2.19-15_amd64.deb, unpacked it and put the libs into a separate path.
In a shell i set the LD_LIBRARY_PATH with
export LD_LIBRARY_PATH=/srv/bin/libc6_2.19

But then i get
$ openscam
Inconsistency detected by ld.so: dl-close.c: 743: _dl_close: Assertion `map->l_init_called' failed!

jcoffland, 2015-04-01:
Sorry you are having so much trouble.  Lots of people are using OpenSCAM in Windows.  I just tested this myself yesterday.  Did your son see any errors you could report?

Re: Debian testing with X on NVidia.  I've got the same setup.  You should install the proprietary NVidia drivers from non-free.  The Open-Source driver has problems.  You should be able to just install the ```nvidia-driver``` meta package.

karstengit, 2015-04-01:
A sorry is not needed - you are writing this software opensource!
Sometimes there are special problems with the users ... :-)

I will ask for the problems in windows.
As i remember the g-code was loaded and then nothing happens.