CauldronDevelopmentLLC/CAMotics #130: Buildung from sources failed in Debian stable

karstengit, 2015-04-01:
Please refer to https://github.com/CauldronDevelopmentLLC/OpenSCAM/issues/129

I tried to build OpemSCAM from source as described.
The first part with cbang works. I attached the result, compiled for AMD64.
[image]

The compiling of OpenSCAM failed with many errors.
I attached the output.
[image]

jcoffland, 2015-04-01:
It looks like you built C! with out libv8 but then installed it later.  You need to rebuild C!.  Unfortunately, once scons has detected that it cannot find libv8 it caches the result and continues to think the library is missing.  Perhaps the easiest way is to just do a full rebuild of C!.  Make sure the configuration phase for C! says, ```Checking for C library v8... (cached) yes``` with or with out the ```(cached)``` part.

You might also run in to problems with not having a new enough version of libv8.  Let me know how it goes.

karstengit, 2015-04-01:
I am sorry - i never have to build C to compile something.
I would say i have only novice knowledge and i am learning ... ;-)

C needs to have a library included?

Today i will give up.
For the future i need a newer Linux distro to work with newer software.
That's the better solution of course.
But it would be fine when software works with actual stable distro's on the other hand ...

jcoffland, 2015-04-01:
C! = cbang.  ```!``` can be read as "bang".

karstengit, 2015-04-02:
The compiler must explode !?

jcoffland, 2015-04-02:
Sometimes it does.

karstengit, 2015-04-02:
Do i need a cleaning program afterwards? :-)

closed 2015-08-23 by jcoffland