CauldronDevelopmentLLC/CAMotics #131: Automatic Process Planning

BillBob12, 2015-04-22:
I've been working on an automatic tool path generating concept and was wondering whether this is something you'd be interested in developing

WJHildreth, 2015-04-22:
Joesph, 

Is this a question in general to the group or to me specifically? 

Joe 

Joe Hildreth 
IT Manager 
Three Rivers Hospital 
451 Hwy 13 South 
Waverly, TN 37185 
(931) 296-0217 (office) 
(931) 622-3130 (cell) 

-----

BillBob12, 2015-04-22:
I guess it was intended to the group in general. 
I'm pretty fresh to github so sorry if i fucked it up.
I'm just coming to the end of this project and was wondering if other people were looking into it too.

WJHildreth, 2015-04-22:
No worries, 

I know we talked in the past about me helping out in some areas. I have been slammed the last several months so have been of no use to you or the project. I have just been lurking, but I think the next 8 weeks or so will finish some of the big stuff I have going on and I hope to reintroduce myself back into the project. 

I am new to guthub myself, so I didn't fully understand if this was a group message or a message to me. Mea Culpa. 

Again, if you need something from me, I am more than happy to help in whatever capacity I can. 

Joe 

Joe Hildreth 
IT Manager 
Three Rivers Hospital 
451 Hwy 13 South 
Waverly, TN 37185 
(931) 296-0217 (office) 
(931) 622-3130 (cell) 

-----

jcoffland, 2015-04-22:
@WJHildreth these messages go to everyone subscribed to GitHub notification for OpenSCAM.  If you want to unsubscribe then got to GitHub and click the **Unsubscribe** button on the OpenSCAM project page or edit your notification preferences.

jcoffland, 2015-04-22:
@BillBob12 tell me more about what you're doing.  Do you have any examples and/or code?

BillBob12, 2015-04-23:
I've been using genetic algorithms to generate 2D toolpaths for lathes, written in Python with DEAP. It's not very advanced but it's piqued my interest. I submit my report on it on Wednesday so after then i'll be free to show the bits and bobs if you wanna see. 
But what approaches are you using for toolpath generation? I couldn't find much about what you're doing

jcoffland, 2015-04-23:
Still in alpha but check out tplang.org.  The approach is very direct.  How do genetic algorithms help with toolpath generation?

jCandlish, 2015-04-23:
Hi jocffland

Thanks for OpenSCAM, and making it opensource.  I took a look to your tpl stuff and thought it was pretty neat.  The idea of of canonical motions is fundamental and implementing it in JavaScript make the library available to web clients.

As far as motion planning is concerned, Rice Univeristy has made available their [Open Motion Planning Library](http://ompl.kavrakilab.org/).  To wit:

*OMPL, the Open Motion Planning Library, consists of many state-of-the-art sampling-based motion planning algorithms. OMPL itself does not contain any code related to, e.g., collision checking or visualization. This is a deliberate design choice, so that OMPL is not tied to a particular collision checker or visualization front end.*

Although it offers no collision detection (the other NP-hard part of tool path planning) it is a very interesting codebase, well documented with references for collision detection candidates.

I just thought that I would offer that in case you or others were not aware.

Cheers
jCandlish
.

BillBob12, 2015-04-23:
i was using GAs to find a toolpath for a given profile.
So the input is a bunch of lines making up the profile of a part and then the GA throws a bunch of toolpaths at it 'till one fits, then has a go at optimising them for machine time or whatever. Or at least that's the idea. It doesn't really work yet but there's a lot of promising work being done about the place
What's the deal TPL? seems just like Gcode but with words instead of the G & M codes.

jcoffland, 2015-04-23:
@jCandlish OMPL looks very interesting.  I imagine we can use it at some point.  However, it will take more research to get from the problems that OMPL solves, namely how do I get from point A to point B with out collisions, to machining problems like, how do I remove remove material from a workpiece in order to yield the desired final product.  Clearly these problems are related but getting from one to the other is non-trivial.

jcoffland, 2015-04-23:
@BillBob12 TPL does act as a more readable replacement for GCode, which is sorely needed.  However, it also adds all the language constructs provided by Javascript such as loops, conditionals, functions, etc.  The version of GCode supported by LinuxCNC does supply some of these constructs but in a very syntactically ugly way (IMHO) on top of GCode.

Aside from better code readability and more powerful language constructs, TPL will also provide basic CAM functions such as pocketing and offsetting, the ability to read and operate on DXF and SVG files and more advanced functions such as the ability to cut fonts from strings of text, drop cutter algorithms and many other planed features.  So, whereas GCode is mainly a machine control language, TPL is also a tool path design language with built-in CAM operations.

You might ask why not just use Python or some other language.  Although Python is a great language, I wanted TPL to be more accessible and to stand on it's own.  I don't want it to carry all the weight of a general purpose software system like Python (or even node.js for that matter).  TPL will focus specifically on tool path generation.

I don't know if you noticed but TPL outputs GCode so it remains compatible with many existing CNC systems.  It is feasible that TPL could output other control languages understood by CNC machines, such as HPGL, as well.

jcoffland, 2015-07-01:
I need some more concrete details before I can act on this issue.  I am closing it for now.  If you have specifics about what you would like to implement please describe them in detail and reopen this ticket.

closed 2015-07-01 by jcoffland