CauldronDevelopmentLLC/CAMotics #133: build of dev branch fails

muetze, 2015-05-03:
Hi,
I try to build the dev branch of OpenSCAM, but it fails due to missing  TriangleMesh.h. Master branch builds fine.

scons: Building targets ...
build_function(["build/build_info.cpp"], [])
g++ -o build/openscam/contour/ElementSurface.o -c -std=gnu++98 -O3 -funroll-loops -mfpmath=sse -ffast-math -fno-unsafe-math-optimizations -msse2 -DNDEBUG -D_REENTRANT -DHAVE_EXPAT -DHAVE_PTHREADS -DHAVE_LIBSQLITE -DHAVE_OPENSSL -DHAVE_CBANG -DUSING_CBANG -DGLEW_STATIC -DQT_CORE_LIB -DQT_GUI_LIB -DQT_OPENGL_LIB -DQT_SHARED -Icbang/src -Icbang/include -I/usr/include/qt4 -I/usr/include/qt4/QtOpenGL -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtCore -I/usr/include/freetype2 -Isrc -Ibuild build/openscam/contour/ElementSurface.cpp
build/openscam/contour/ElementSurface.cpp:23:26: fatal error: TriangleMesh.h: No such file or directory
 #include "TriangleMesh.h"
                          ^
compilation terminated.
scons: *** [build/openscam/contour/ElementSurface.o] Error 1
scons: building terminated because of errors.
Makefile:2: recipe for target 'all' failed
make: *** [all] Error 2

I had an older dev-build running and liked it a lot, especially the integrated editor.
Cheers, muetze

jcoffland, 2015-05-04:
Thanks for reporting this.  I forgot to check in some files.  Should work now.

closed 2015-05-04 by jcoffland

muetze, 2015-05-04:
Works now. 
Thanks!

jcoffland, 2015-05-04:
No problem.