CauldronDevelopmentLLC/CAMotics #135: Compile Error with Debian Wheezy again

wolfkarst, 2015-05-09:
I am using Debian Wheezy with Backports
libv8 is 3.14.5.8-8bpo
gcc is 4.4 or 4.6 or 4.7 (no difference)
I work strict like readme.md with fresh Dirs.
CBang compiles without any error
Openscam compiles a lot till this:

ar rc libclipper.a build/clipper/clipper.o
ranlib libclipper.a
g++ -o openscam -Wl,-S -Wl,-x -pthread build/openscam.o build/qrc_openscam.o libOpenSCAM.a libclipper.a -L/opt/openscam/cbang/lib -lcairo -lv8 -lGLU -lGL -lpng -lz -lfreetype -lcbang -lv8 -lssl -lcrypto -lboost_regex -lboost_filesystem -lboost_system -lboost_iostreams -levent -lsqlite3 -ldl -lexpat -lbz2 -lz -lpthread -lQtOpenGL -lQtGui -lQtCore
libOpenSCAM.a(DXFLibrary.o): In function `tplang::DXFLibrary::cutCB(cb::js::Arguments const&)':
DXFLibrary.cpp:(.text+0x5): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase()'
libOpenSCAM.a(DXFLibrary.o): In function `tplang::DXFLibrary::add(cb::js::ObjectTemplate&)':
DXFLibrary.cpp:(.text+0xbc4): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(int)'
DXFLibrary.cpp:(.text+0xc0f): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(int)'
DXFLibrary.cpp:(.text+0xc5a): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(int)'
DXFLibrary.cpp:(.text+0xca5): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(int)'
DXFLibrary.cpp:(.text+0xcf0): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(int)'
libOpenSCAM.a(DXFLibrary.o): In function `tplang::DXFLibrary::openCB(cb::js::Arguments const&)':
DXFLibrary.cpp:(.text+0x14eb): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::createArray(unsigned int)'
DXFLibrary.cpp:(.text+0x19d1): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::createArray(unsigned int)'
DXFLibrary.cpp:(.text+0x1a30): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(double)'
DXFLibrary.cpp:(.text+0x1a64): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x1a95): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(double)'
DXFLibrary.cpp:(.text+0x1ac9): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x1af8): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(double)'
DXFLibrary.cpp:(.text+0x1b2c): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x1b54): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(int)'
DXFLibrary.cpp:(.text+0x1b88): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x1bb6): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(cb::js::ValueBase<v8::Handle<v8::Value> > const&)'
DXFLibrary.cpp:(.text+0x1bce): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(unsigned int, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x1c0d): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(cb::js::ValueBase<v8::Handle<v8::Value> > const&)'
DXFLibrary.cpp:(.text+0x1c3e): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x1c71): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(int)'
DXFLibrary.cpp:(.text+0x1ca2): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x1ccd): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(cb::js::ValueBase<v8::Handle<v8::Value> > const&)'
DXFLibrary.cpp:(.text+0x1ce3): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(unsigned int, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x1dac): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(double)'
DXFLibrary.cpp:(.text+0x1de0): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x1e0b): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(double)'
DXFLibrary.cpp:(.text+0x1e3f): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x1e6a): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(double)'
DXFLibrary.cpp:(.text+0x1e9e): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x1ecc): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(cb::js::ValueBase<v8::Handle<v8::Value> > const&)'
DXFLibrary.cpp:(.text+0x1efd): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x1f28): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(double)'
DXFLibrary.cpp:(.text+0x1f59): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x1f84): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(double)'
DXFLibrary.cpp:(.text+0x1fb5): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x1fe0): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(double)'
DXFLibrary.cpp:(.text+0x2011): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x2094): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(double)'
DXFLibrary.cpp:(.text+0x20c8): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x20f3): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(double)'
DXFLibrary.cpp:(.text+0x2127): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x2152): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(double)'
DXFLibrary.cpp:(.text+0x2186): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x21b4): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(cb::js::ValueBase<v8::Handle<v8::Value> > const&)'
DXFLibrary.cpp:(.text+0x21e5): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x2220): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(double)'
DXFLibrary.cpp:(.text+0x2254): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x227f): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(double)'
DXFLibrary.cpp:(.text+0x22b3): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x22de): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(double)'
DXFLibrary.cpp:(.text+0x2312): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)'
DXFLibrary.cpp:(.text+0x2340): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(cb::js::ValueBase<v8::Handle<v8::Value> > const&)'
[... cut ...]
/opt/openscam/cbang/lib/libcbang.a(StdLibrary.o): In function `cb::js::VoidMethodCallback<cb::js::StdLibrary>::operator()(cb::js::Arguments const&)':
StdLibrary.cpp:(.text._ZN2cb2js18VoidMethodCallbackINS0_10StdLibraryEEclERKNS0_9ArgumentsE[_ZN2cb2js18VoidMethodCallbackINS0_10StdLibraryEEclERKNS0_9ArgumentsE]+0x2b): undefined reference to `cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase()'
collect2: error: ld returned 1 exit status
scons: *** [openscam] Error 1
scons: building terminated because of errors.

jcoffland, 2015-05-09:
These errors are caused by compiling cbang with out libv8 support.  libv8 is optional is cbang but required for OpenSCAM.  You need to make sure it says it's detected during sons' configuration phase.

Another problem is that the cbang build system often fails to recognized that you've added libv8 to the system once it has already configured itself once.  It caches the old value.  The sure fire way to make sure it builds correctly is either to dump the cbang build and start over or remove the scons data from cbang like this:

    cd cbang
    rm -rf .sconf_temp/ .sconsign.dblite

Then rebuild cbang and OpenSCAM.

wolfkarst, 2015-05-09:
I have made 
cd cbang
rm -rf .sconf_temp/ .sconsign.dblite

While Compiling Cbang :
Checking for C++ header file v8.h... yes
Checking for C library v8... yes

But later in the Compile-Process of CBang no "v8" apears ...

Checking export CBANG_HOME ... is OK

cd OpenSCAM
rm -rf .sconf_temp/ .sconsign.dblite

While Compiling OpenSCAM :
Checking for C++ header file v8.h... yes
Checking for C library v8... yes
Checking for C++ header file v8.h... yes
Checking for C library v8... yes                              // Yes it appears twice

And the Problem is still there.

By the way: can it be a Problem with 64-Bit Linux ? Do I need the 386er Libary too ?

wolfkarst, 2015-05-09:
Libv8 is placed in /usr/lib :
lrwxrwxrwx 1 root root      15 Mai 28  2014 /usr/lib/libv8.so -> libv8.so.3.14.5
-rw-r--r-- 1 root root 4207440 Mai 28  2014 /usr/lib/libv8.so.3.14.5

and v8.h is placed in /usr/include :
-rw-r--r-- 1 root root  15076 Mär  4  2013 /usr/include/v8-debug.h
-rw-r--r-- 1 root root 150785 Mär  4  2013 /usr/include/v8.h
-rw-r--r-- 1 root root   4591 Mär  4  2013 /usr/include/v8-preparser.h
-rw-r--r-- 1 root root  18630 Mär  4  2013 /usr/include/v8-profiler.h
-rw-r--r-- 1 root root   2140 Mär  4  2013 /usr/include/v8stdint.h
-rw-r--r-- 1 root root   3352 Mär  4  2013 /usr/include/v8-testing.h

jcoffland, 2015-05-09:
I'm building on 64-bit Debian *testing* with libv8-4.14.5.8-8.1 and it's working fine for me.  I'm not sure what's going on here.  Is it possible you have another libv8 or cbang somewhere else on your system that's getting picked up during the linking phase?

All of the errors refer to ```cb::js::ValueBase<v8::Handle<v8::Value> >``` which indicates that this symbol is not present in your libcbang.a.  What is the output of the following command?:

    cd cbang
    nm -C lib/libcbang.a | grep 'cb::js::ValueBase<v8::Handle<v8::Value> >' | wc -l

I get ```411```.  If you see ```0``` then the symbol is truly missing from your cbang.  If you see a positive number then something else is wrong.

wolfkarst, 2015-05-09:
/opt/openscam/cbang$ nm -C lib/libcbang.a | grep 'cb::js::ValueBase<v8::Handle<v8::Value> >' | wc -l
78

wolfkarst, 2015-05-09:
My libv8 is from debian wheezy Backports : 3.14.5.8-8-bpo70+1

"find / -name libv8.so" reports only
/usr/lib/libv8.so
"find / -name cbang" reports only
/opt/openscam/cbang
/opt/openscam/cbang/config/cbang
/opt/openscam/cbang/build/cbang
/opt/openscam/cbang/src/cbang
these are the correct locations.

Like i mentioned above:
Is it normal, that no "v8" apears in the Compiling-Output of CBang after the Checking ?

jcoffland, 2015-05-10:
It's strange that your libcbang.a only has 78 instances of the undefined symbol's class where mine has 411.  What's the output of this command?:

    nm -C lib/libcbang.a | grep 'cb::js::ValueBasev8::Handle<v8::Value >'

It is normal that the compiling output does not mention libv8.  It's including from ```/usr/include``` which is the default include path.  Cbang is a library and does not need to link against ```libv8```.

wolfkarst, 2015-05-10:
This output is empty (I think you missed some "<"
The Output of nm -C lib/libcbang.a | grep 'cb::js::ValueBase<v8::Handle<v8::Value> >' is : 

                 U cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(v8::Handle<v8::Value> const&)
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(cb::js::ValueBase<v8::Handle<v8::Value> > const&)
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase()
0000000000000000 W cb::OrderedDict<cb::js::ValueBase<v8::Handle<v8::Value> >, std::string>::get(unsigned long) const
0000000000000000 W cb::OrderedDict<cb::js::ValueBase<v8::Handle<v8::Value> >, std::string>::indexOf(std::string const&) const
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::assertDefined() const
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::getOwnPropertyNames() const
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::get(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >) const
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::get(unsigned int) const
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::has(std::string const&) const
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::length() const
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::toString() const
0000000000000000 W cb::OrderedDict<cb::js::ValueBase<v8::Handle<v8::Value> >, std::string>::~OrderedDict()
0000000000000000 W cb::OrderedDict<cb::js::ValueBase<v8::Handle<v8::Value> >, std::string>::~OrderedDict()
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(cb::js::ValueBase<v8::Handle<v8::Value> > const&)
0000000000000000 W void std::_Destroy_aux<false>::__destroy<std::pair<std::string, cb::js::ValueBase<v8::Handle<v8::Value> > >*>(std::pair<std::string, cb::js::ValueBase<v8::Handle<v8::Value> > >*, std::pair<std::string, cb::js::ValueBase<v8::Handle<v8::Value> > >*)
0000000000000000 W std::vector<std::pair<std::string, cb::js::ValueBase<v8::Handle<v8::Value> > >, std::allocator<std::pair<std::string, cb::js::ValueBase<v8::Handle<v8::Value> > > > >::~vector()
0000000000000000 W std::vector<std::pair<std::string, cb::js::ValueBase<v8::Handle<v8::Value> > >, std::allocator<std::pair<std::string, cb::js::ValueBase<v8::Handle<v8::Value> > > > >::~vector()
                 U cb::js::ValueBase<v8::Persistent<v8::Value> >::ValueBase(cb::js::ValueBase<v8::Handle<v8::Value> > const&)
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::createObject()
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> >)
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::call(cb::js::ValueBase<v8::Handle<v8::Value> >, cb::js::ValueBase<v8::Handle<v8::Value> >)
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(v8::Handle<v8::Value> const&)
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(cb::js::ValueBase<v8::Handle<v8::Value> > const&)
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(std::string const&)
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(int)
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::getOwnPropertyNames() const
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::get(std::string const&) const
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::length() const
0000000000000120 T cb::js::ObjectTemplate::set(cb::js::Signature const&, cb::js::ValueBase<v8::Handle<v8::Value> > (*)(cb::js::Arguments const&))
0000000000000060 T cb::js::ObjectTemplate::set(std::string const&, cb::js::ValueBase<v8::Handle<v8::Value> > const&)
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(v8::Handle<v8::Value> const&)
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(v8::Handle<v8::Value> const&)
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::toString() const
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(v8::Handle<v8::Value> const&)
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(cb::js::ValueBase<v8::Handle<v8::Value> > const&)
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(std::string const&)
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(bool)
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(double)
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(int)
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase(unsigned int)
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::ValueBase()
0000000000000000 W cb::OrderedDict<cb::js::ValueBase<v8::Handle<v8::Value> >, std::string>::get(unsigned long) const
0000000000000000 W cb::OrderedDict<cb::js::ValueBase<v8::Handle<v8::Value> >, std::string>::keyAt(unsigned long) const
                 U cb::js::ValueBase<v8::Handle<v8::Value> >::toString() const
0000000000000000 W void std::_Destroy_aux<false>::__destroy<std::pair<std::string, cb::js::ValueBase<v8::Handle<v8::Value> > >*>(std::pair<std::string, cb::js::ValueBase<v8::Handle<v8::Value> > >*, std::pair<std::string, cb::js::ValueBase<v8::Handle<v8::Value> > >*)
0000000000000000 W std::pair<std::string, cb::js::ValueBase<v8::Handle<v8::Value> > >::~pair()
0000000000000000 W std::pair<std::string, cb::js::ValueBase<v8::Handle<v8::Value> > >::~pair()
[... cut ...]
0000000000000000 W cb::js::ValueBase<v8::Handle<v8::Value> >::isString() const
0000000000000000 W cb::js::ValueBase<v8::Handle<v8::Value> >::isUint32() const
0000000000000000 W cb::js::ValueBase<v8::Handle<v8::Value> >::toNumber() const
0000000000000000 W cb::js::ValueBase<v8::Handle<v8::Value> >::toUint32() const
0000000000000000 W cb::js::ValueBase<v8::Handle<v8::Value> >::isBoolean() const
0000000000000000 W cb::js::ValueBase<v8::Handle<v8::Value> >::toBoolean() const
0000000000000000 W cb::js::ValueBase<v8::Handle<v8::Value> >::toInteger() const

jcoffland, 2015-05-11:
Your library is missing 308 of the symbols matching ```cb::js::ValueBase<v8::Handle<v8::Value> >``` that mine has.   I don't see any pattern.  This is really strange.  I've never seen this before.

What's the output of this command?:

    readelf -a -W lib/libcbang.a | grep Class: | sort | uniq

I get this:

      Class:                             ELF64

wolfkarst, 2015-05-11:
```  Class:                             ELF64```

Can you make a .deb-Package for Debian wheezy? The Deb in the Downloads needs a higher Glibc.My System has only 2.13 .

wolfkarst, 2015-05-11:
Whats the Size of your libv8.so ? mine in 4.2 MB

jcoffland, 2015-05-11:
Mine is 4.2MiB too, but the missing symbols are from cbang not libv8.  Have you tried the LinuxCNC build?  It's for an older Ubuntu.

wolfkarst, 2015-05-11:
I have  tried it now. One Problem: It needs libboost_iostreams.so in i386, but if i change it from x64 to i386 then it will kill my apt-system ...

wolfkarst, 2015-05-11:
Versions for Ubuntu 10.04 up to 13.10 are equal to Debian Wheezy and Squeeze and should work with my system. LINUX-CNC for 10.04 should be fine, but is for i386 and not for x64 and mixing Bit-Level is not very stabil at Squeeze-Level.(10.04-11.04)

jcoffland, 2015-05-12:
You're right the LinuxCNC build is 32-bit.  Sorry about that.  I don't have a 64-bit build system setup that meets your needs.  I could setup a VM but I don't have time for that right now.  I'm working on getting a Website ready for beta testing.

koppi, 2015-05-16:
Do you have an easy fix? I just got into the same issue on Debian Wheezy with backports enabled.

jcoffland, 2015-05-17:
@koppi, sorry I don't know what's going on here.  I'll have to setup a VM with Wheezy w/ backports to debug this.

koppi, 2015-05-17:
Are you using docker.io – I could help out with a Dockerfile..

koppi, 2015-05-17:
Here's the Dockerfile:
```bash
FROM debian:wheezy

RUN apt-get update && DEBIAN_FRONTEND=noninteractive apt-get install -y dpkg-dev apt-utils; \
    echo "deb http://ftp.de.debian.org/debian/ wheezy-backports main non-free contrib" >> /etc/apt/sources.list; \
    apt-get update
        
WORKDIR /root

RUN apt-get install -y \
        build-essential fakeroot dpkg-dev dh-autoreconf \
        scons build-essential libbz2-dev zlib1g-dev \
        libexpat1-dev libssl-dev libboost-iostreams-dev libboost-system-dev
RUN apt-get install -y \
        libboost-filesystem-dev libboost-regex-dev libsqlite3-dev libv8-dev \
        qt4-dev-tools libqt4-dev libqt4-opengl-dev libcairo2-dev git

RUN git clone https://github.com/CauldronDevelopmentLLC/cbang
RUN git clone https://github.com/CauldronDevelopmentLLC/OpenSCAM
RUN scons -C cbang; \
    export CBANG_HOME=$PWD/cbang; \
    cd OpenSCAM; \
    scons -j 5 && \
    scons package
        
VOLUME /output
CMD cp *.deb /output
```

jcoffland, 2015-05-18:
@koppi, thank you that was a big help.  I'm now able to reproduce the error.  I'll debug and see what I can figure out.

jcoffland, 2015-05-18:
Ok, I've fixed it.  Thanks for the help.

closed 2015-05-18 by jcoffland

koppi, 2015-05-18:
Were is the fix? – I do not see any change / update of the repo.

jcoffland, 2015-05-18:
https://github.com/CauldronDevelopmentLLC/cbang/commit/d917cc00811faf63596ffd8d7788bd3e3262032f

koppi, 2015-05-18:
Thanks, I'm getting another error:

```
scons: Building targets ...
g++ -o build/openscam.o -c -std=gnu++98 -O3 -funroll-loops -mfpmath=sse -ffast-math -fno-unsafe-math-optimizations -msse2 -DNDEBUG -D_REENTRANT -DHAVE_EXPAT -DHAVE_PTHREADS -DHAVE_LIBSQLITE -DHAVE_OPENSSL -DHAVE_CBANG -DUSING_CBANG -DGLEW_STATIC -DQT_CORE_LIB -DQT_GUI_LIB -DQT_OPENGL_LIB -DQT_SHARED -I/home/koppi/cbang/src -I/home/koppi/cbang/include -I/usr/include/qt4 -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtOpenGL -I/usr/include/freetype2 -Isrc -Ibuild build/openscam.cpp
build_function(["build/build_info.cpp"], [])
g++ -o build/build_info.o -c -std=gnu++98 -O3 -funroll-loops -mfpmath=sse -ffast-math -fno-unsafe-math-optimizations -msse2 -DNDEBUG -D_REENTRANT -DHAVE_EXPAT -DHAVE_PTHREADS -DHAVE_LIBSQLITE -DHAVE_OPENSSL -DHAVE_CBANG -DUSING_CBANG -DGLEW_STATIC -DQT_CORE_LIB -DQT_GUI_LIB -DQT_OPENGL_LIB -DQT_SHARED -I/home/koppi/cbang/src -I/home/koppi/cbang/include -I/usr/include/qt4 -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtOpenGL -I/usr/include/freetype2 -Isrc -Ibuild build/build_info.cpp
g++ -o build/openscam/cutsim/CutSim.o -c -std=gnu++98 -O3 -funroll-loops -mfpmath=sse -ffast-math -fno-unsafe-math-optimizations -msse2 -DNDEBUG -D_REENTRANT -DHAVE_EXPAT -DHAVE_PTHREADS -DHAVE_LIBSQLITE -DHAVE_OPENSSL -DHAVE_CBANG -DUSING_CBANG -DGLEW_STATIC -DQT_CORE_LIB -DQT_GUI_LIB -DQT_OPENGL_LIB -DQT_SHARED -I/home/koppi/cbang/src -I/home/koppi/cbang/include -I/usr/include/qt4 -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtOpenGL -I/usr/include/freetype2 -Isrc -Ibuild build/openscam/cutsim/CutSim.cpp
In file included from src/tplang/TPLContext.h:28:0,
                 from build/openscam/cutsim/CutSim.cpp:28:
/home/koppi/cbang/src/cbang/js/Environment.h:39:27: fatal error: ConsoleModule.h: No such file or directory
compilation terminated.
g++ -o build/openscam/qt/ToolPathThread.o -c -std=gnu++98 -O3 -funroll-loops -mfpmath=sse -ffast-math -fno-unsafe-math-optimizations -msse2 -DNDEBUG -D_REENTRANT -DHAVE_EXPAT -DHAVE_PTHREADS -DHAVE_LIBSQLITE -DHAVE_OPENSSL -DHAVE_CBANG -DUSING_CBANG -DGLEW_STATIC -DQT_CORE_LIB -DQT_GUI_LIB -DQT_OPENGL_LIB -DQT_SHARED -I/home/koppi/cbang/src -I/home/koppi/cbang/include -I/usr/include/qt4 -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtOpenGL -I/usr/include/freetype2 -Isrc -Ibuild build/openscam/qt/ToolPathThread.cpp
scons: *** [build/openscam/cutsim/CutSim.o] Error 1
scons: building terminated because of errors.
koppi@x200:~/OpenSCAM$ git pull
Already up-to-date.
```

jcoffland, 2015-05-19:
Sorry about that. I made some more changes later and forgot to commit a couple of files.  It should be ok now.

koppi, 2015-05-19:
@jcoffland thanks, openscam now compiles and runs. What about adding a .travis.yml?

Here's an example https://github.com/openscad/openscad/blob/master/.travis.yml

koppi, 2015-05-19:
Works great, thanks!

jcoffland, 2015-05-19:
Glad to hear it.  Thanks for the help.  I'll look into travisci.  I've avoided it in the past because it didn't support Windows builds.  However, I hear that's changing.  I have a Buildbot setup which builds releases for me on Debian, CentOS, Ubuntu, OSX & Windows.  It can be difficult to maintain though.

jcoffland, 2015-05-19:
@koppi [Travis CI build](https://travis-ci.org/CauldronDevelopmentLLC/OpenSCAM) is working.

koppi, 2015-05-20:
Excellent! Seems like travis cs currently supports Linux and OSX, hope they
will add Windows someday sooner or later, too.

jcoffland, 2015-05-20:
@koppi, they are not accepting any new requests to use OSX, as far as I can tell.  So really it's only Linux for now.  I'd like to see a similar service which worked with all three OSes and only in Docker like containers.  Then we'd really have something.

koppi, 2015-05-20:
Did you have a look into Vagrantfiles?

jcoffland, 2015-05-21:
I've played a bit with Vagrant.  I've got my own VMs for Windows and Linux.  Are their any services like Travis CI that use vagrant and support Windows and OSX as well as Linux that are free for Open-Source?  All I want is the holy grail of CI.

koppi, 2015-06-03:
Did you hear of https://www.appveyor.com/

jcoffland, 2015-06-03:
No, I had not.  Looks interesting but it would still be nice to have a solution that works with all three platforms.  The Appveyor configuration looks similar to Travis CI though.  Nice to see that it's free for Open-Source.  I might give it a try.