CauldronDevelopmentLLC/CAMotics #136: Install fails on OS X Yosemite

LeHonk, 2015-05-17:
I've tried to install on Mac osx 10.10.3 but it fails with some less than helpful error message
"Something went wrong"

But rerun with logging output and there seems to be something wrong with the `postflight` script.
I know to little about OS X installer to debug further but include the log output and hope it helps.

Installer logfile:
```
May 17 18:30:01 iMac.local Installer[94716]: LSExceptions [0x6000000b50c0] loaded
May 17 18:30:01 iMac.local Installer[94716]: @(#)PROGRAM:Install  PROJECT:Install-920
May 17 18:30:01 iMac.local Installer[94716]: @(#)PROGRAM:Installer  PROJECT:Installer-815
May 17 18:30:01 iMac.local Installer[94716]: Hardware: iMac10,1 @ 3.06 GHz (x 2), 4096 MB RAM
May 17 18:30:01 iMac.local Installer[94716]: Running OS Build: Mac OS X 10.10.3 (14D136)
May 17 18:30:01 iMac.local Installer[94716]: Env: TMPDIR=/var/folders/x0/nrs9wstj023ggj7pngr2nh580000gn/T/
May 17 18:30:01 iMac.local Installer[94716]: Env: __CF_USER_TEXT_ENCODING=0x1F5:0x0:0x7
May 17 18:30:01 iMac.local Installer[94716]: Env: HOME=/Users/tomasholmqvist
May 17 18:30:01 iMac.local Installer[94716]: Env: SHELL=/bin/bash
May 17 18:30:01 iMac.local Installer[94716]: Env: Apple_PubSub_Socket_Render=/private/tmp/com.apple.launchd.TMmb1A15zq/Render
May 17 18:30:01 iMac.local Installer[94716]: Env: SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.HanCbIwfqS/Listeners
May 17 18:30:01 iMac.local Installer[94716]: Env: PATH=/usr/bin:/bin:/usr/sbin:/sbin
May 17 18:30:01 iMac.local Installer[94716]: Env: LOGNAME=tomasholmqvist
May 17 18:30:01 iMac.local Installer[94716]: Env: DISPLAY=/private/tmp/com.apple.launchd.dU3nOArwlQ/org.macosforge.xquartz:0
May 17 18:30:01 iMac.local Installer[94716]: Env: XPC_SERVICE_NAME=com.apple.installer.1452
May 17 18:30:01 iMac.local Installer[94716]: Env: USER=tomasholmqvist
May 17 18:30:01 iMac.local Installer[94716]: Env: XPC_FLAGS=0x0
May 17 18:30:01 iMac.local Installer[94716]: Open-Source Simulation & Computer Aided Machining  Installation Log
May 17 18:30:01 iMac.local Installer[94716]: Opened from: /Users/tomasholmqvist/Downloads/openscam_0.2.5_x86_64.pkg
May 17 18:30:02 iMac.local Installer[94716]: Product archive /Users/tomasholmqvist/Downloads/openscam_0.2.5_x86_64.pkg trustLevel=100
May 17 18:30:11 iMac.local Installer[94716]: LSExceptions [0x6000000b50c0] unloaded
May 17 18:30:51 iMac.local Installer[94716]: InstallerStatusNotifications plugin loaded
May 17 18:30:55 iMac.local Installer[94716]: ================================================================================
May 17 18:30:55 iMac.local Installer[94716]: User picked Standard Install
May 17 18:30:55 iMac.local Installer[94716]: Choices selected for installation:
May 17 18:30:55 iMac.local Installer[94716]: 	Installera: "Open-Source Simulation & Computer Aided Machining"
May 17 18:30:55 iMac.local Installer[94716]: 	Installera: "(null)"
May 17 18:30:55 iMac.local Installer[94716]: 		openscam_0.2.5_x86_64.pkg#OpenSCAM.pkg : org.openscam.pkg : 0.2.5
May 17 18:30:55 iMac.local Installer[94716]: ================================================================================
May 17 18:30:55 iMac.local Installer[94716]: It took 0.00 seconds to summarize the package selections.
May 17 18:30:55 iMac.local Installer[94716]: -[IFDInstallController(Private) _buildInstallPlan]: location = file://localhost
May 17 18:30:55 iMac.local Installer[94716]: -[IFDInstallController(Private) _buildInstallPlan]: file://localhost/Users/tomasholmqvist/Downloads/openscam_0.2.5_x86_64.pkg#OpenSCAM.pkg
May 17 18:30:55 iMac.local Installer[94716]: Set authorization level to root for session
May 17 18:31:02 iMac.local runner[94720]: Administrator authorization granted.
May 17 18:31:02 iMac.local Installer[94716]: Will use PK session
May 17 18:31:02 iMac.local Installer[94716]: Starting installation:
May 17 18:31:02 iMac.local Installer[94716]: Configuring volume "Seagate 1TB HD"
May 17 18:31:02 iMac.local Installer[94716]: Preparing disk for local booted install.
May 17 18:31:02 iMac.local Installer[94716]: Free space on "Seagate 1TB HD": 835,68 GB (835679002624 bytes).
May 17 18:31:02 iMac.local Installer[94716]: Create temporary directory "/var/folders/x0/nrs9wstj023ggj7pngr2nh580000gn/T//Install.94716WfPM9B"
May 17 18:31:02 iMac.local Installer[94716]: IFPKInstallElement (1 packages)
May 17 18:31:02 iMac.local Installer[94716]: Using authorization level of root for IFPKInstallElement
May 17 18:31:02 iMac.local installd[561]: PackageKit: Adding client PKInstallDaemonClient pid=94716, uid=501 (/System/Library/CoreServices/Installer.app/Contents/MacOS/Installer)
May 17 18:31:02 iMac.local Installer[94716]: PackageKit: Enqueuing install with framework-specified quality of service (utility)
May 17 18:31:02 iMac.local installd[561]: PackageKit: ----- Begin install -----
May 17 18:31:02 iMac.local installd[561]: PackageKit: request=PKInstallRequest <1 packages, destination=/>
May 17 18:31:02 iMac.local installd[561]: PackageKit: packages=(
	    "PKLeopardPackage <file://localhost/Users/tomasholmqvist/Downloads/openscam_0.2.5_x86_64.pkg#OpenSCAM.pkg>"
	)
May 17 18:31:03 iMac.local installd[561]: PackageKit: Extracting file://localhost/Users/tomasholmqvist/Downloads/openscam_0.2.5_x86_64.pkg#OpenSCAM.pkg (destination=/var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/C/PKInstallSandboxManager/75CB64B3-88F0-4F5D-8B03-B20774F69B04.activeSandbox/Root, uid=0)
May 17 18:31:04 iMac.local installd[561]: PackageKit: prevent user idle system sleep
May 17 18:31:04 iMac.local installd[561]: PackageKit: suspending backupd
May 17 18:31:05 iMac.local installd[561]: PackageKit: Using trashcan path /var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/T/PKInstallSandboxTrash/75CB64B3-88F0-4F5D-8B03-B20774F69B04.sandboxTrash for sandbox /var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/C/PKInstallSandboxManager/75CB64B3-88F0-4F5D-8B03-B20774F69B04.activeSandbox
May 17 18:31:05 iMac.local install_monitor[94725]: Temporarily excluding: /Applications, /Developer, /Library, /System, /bin, /private, /sbin, /usr
May 17 18:31:05 iMac.local installd[561]: PackageKit: Shoving /var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/C/PKInstallSandboxManager/75CB64B3-88F0-4F5D-8B03-B20774F69B04.activeSandbox/Root (2 items) to /
May 17 18:31:05 iMac.local installd[561]: PackageKit: Executing script "./postflight" in /private/tmp/PKInstallSandbox.F9dvb8/Scripts/org.openscam.pkg.aAMdC9
May 17 18:31:07 iMac.local installd[561]: ./postflight: Unable to find application named 'OpenSCAM.app'
May 17 18:31:07 iMac.local install_monitor[94725]: Re-included: /Applications, /Developer, /Library, /System, /bin, /private, /sbin, /usr
May 17 18:31:08 iMac.local installd[561]: PackageKit: releasing backupd
May 17 18:31:08 iMac.local installd[561]: PackageKit: allow user idle system sleep
May 17 18:31:08 iMac.local installd[561]: PackageKit: Install Failed: Error Domain=PKInstallErrorDomain Code=112 "Ett fel inträffade vid körning av skript från paketet ”openscam_0.2.5_x86_64.pkg”." UserInfo=0x7ffa6d06e340 {NSFilePath=./postflight, NSURL=file://localhost/Users/tomasholmqvist/Downloads/openscam_0.2.5_x86_64.pkg#OpenSCAM.pkg, PKInstallPackageIdentifier=org.openscam.pkg, NSLocalizedDescription=Ett fel inträffade vid körning av skript från paketet ”openscam_0.2.5_x86_64.pkg”.} {
	    NSFilePath = "./postflight";
	    NSLocalizedDescription = "Ett fel intr\U00e4ffade vid k\U00f6rning av skript fr\U00e5n paketet \U201dopenscam_0.2.5_x86_64.pkg\U201d.";
	    NSURL = "file://localhost/Users/tomasholmqvist/Downloads/openscam_0.2.5_x86_64.pkg#OpenSCAM.pkg";
	    PKInstallPackageIdentifier = "org.openscam.pkg";
	}
May 17 18:31:08 iMac.local installd[561]: PackageKit: Running idle tasks
May 17 18:31:08 iMac.local Installer[94716]: install:didFailWithError:Error Domain=PKInstallErrorDomain Code=112 "Ett fel inträffade vid körning av skript från paketet ”openscam_0.2.5_x86_64.pkg”." UserInfo=0x6000002f2c80 {NSFilePath=./postflight, NSURL=file://localhost/Users/tomasholmqvist/Downloads/openscam_0.2.5_x86_64.pkg#OpenSCAM.pkg, PKInstallPackageIdentifier=org.openscam.pkg, NSLocalizedDescription=Ett fel inträffade vid körning av skript från paketet ”openscam_0.2.5_x86_64.pkg”.}
May 17 18:31:08 iMac.local installd[561]: PackageKit: Removing client PKInstallDaemonClient pid=94716, uid=501 (/System/Library/CoreServices/Installer.app/Contents/MacOS/Installer)
May 17 18:31:08 iMac.local installd[561]: PackageKit: Done with sandbox removals
May 17 18:31:08 iMac.local Installer[94716]: Install failed: Installeraren stötte på ett fel som förhindrade installationen. Kontakta programtillverkaren och be om assistans.
May 17 18:31:09 iMac.local Installer[94716]: IFDInstallController 10A80 state = 8
May 17 18:31:09 iMac.local Installer[94716]: Displaying 'Install Failed' UI.
May 17 18:31:09 iMac.local Installer[94716]: 'Install Failed' UI displayed message:'Installeraren stötte på ett fel som förhindrade installationen. Kontakta programtillverkaren och be om assistans.'.
```

LeHonk, 2015-05-17:
Install actually successful despite of error message.
The install actually leaves behind a working install of OpenSCAM.
And looking at the script it seems to just execute the newly installed app, but that fails at install time.

Don't know whats going on but it works.

jcoffland, 2015-05-17:
Ok, I've attempted to fix this.  It should work better in the next release.

closed 2017-03-26 by jcoffland