CauldronDevelopmentLLC/CAMotics #138: Fails on simple gcode

ajayre, 2015-05-18:
This gcode cuts a 70mm square groove. Running the simulation produces nothing.

I defined a 3.175mm endmill tool number 5 in OpenSCAM.

( Made using CamBam - http://www.cambam.co.uk )
( 70mmSquare 5/17/2015 12:43:02 PM )
( T5 : 3.175 )
G21 G90 G64 G40
G0 Z3.0
( T5 : 3.175 )
T5 M6
( Profile1 )
G17
M3 S1000
G0 X-1.5875 Y3.22139
G0 Z1.0
G1 F300.0 Z0.0
G1 F800.0 Y0.0 Z-0.28183
G3 X0.0 Y-1.5875 Z-0.5 I1.5875 J0.0
G1 X70.0
G3 X71.5875 Y0.0 I0.0 J1.5875
G1 Y70.0
G3 X70.0 Y71.5875 I-1.5875 J0.0
G1 X0.0
G3 X-1.5875 Y70.0 I0.0 J-1.5875
G1 Y0.0
G3 X0.0 Y-1.5875 I1.5875 J0.0
G0 Z3.0
G0 X-1.5875 Y3.22139
G0 Z0.5
G1 F300.0 Z-0.5
G1 F800.0 Y0.0 Z-0.78183
G3 X0.0 Y-1.5875 Z-1.0 I1.5875 J0.0
G1 X70.0
G3 X71.5875 Y0.0 I0.0 J1.5875
G1 Y70.0
G3 X70.0 Y71.5875 I-1.5875 J0.0
G1 X0.0
G3 X-1.5875 Y70.0 I0.0 J-1.5875
G1 Y0.0
G3 X0.0 Y-1.5875 I1.5875 J0.0
G0 Z3.0
M5
M30

jcoffland, 2015-05-18:
Worked fine for me.

[image]

There are a few ways to make it not appear.  One is if you move the tool position to zero, then run the simulation.  This can be corrected by moving the position slider on the bottom all the way to the right then clicking the run button (the green circular arrow).

Another other way you can make it not appear is by clicking the ```Show uncut workpiece``` button.

Yet, another possibility is creating a tool that is too short to reach the surface.

If none of those appear to be the reason then please reopen this issue and add a screen shot of the whole window.

closed 2015-05-18 by jcoffland

ajayre, 2015-05-18:
Probably the issue is that I couldn't get the workpiece to be in the right position. The endmill was shown all the way through the workpiece to thin air below. I tried adjusting the workpiece size but all I ended up with was a blue block that was the right size but sitting in the endmill half-way up and this huge white bounding box that stretched off into the distance. That made rotating and viewing very difficult so I gave up on that presuming that if the endmill was sitting half-way into the block it would cut anyway.

Even without running the g-code simulation there was a small slot in the workpiece near the endmill at a strange angle, not perpendicular to the workpiece, so I don't know what that was doing there or where it came from.

I didn't change any settings, I only started the app, added my endmill, opened the g-code file, tried adjusting the size of the workpiece then hit the button to re-run the simulation and waited.

jcoffland, 2015-05-18:
I'm guessing that modifying the workpiece was the problem.  I opened the file, made no changes at all and it worked as expected.

ajayre, 2015-05-18:
Hmmm... strange. As I mentioned all I did was open the gcode file and the default workpiece (without making any changes) shows it "floating" half-way up the endmill. That's why I tried to adjust the size of it.

jcoffland, 2015-05-18:
Ok, now I understand.  Your running into this issue #105.  It's fixed in the version I'm using.  I'm getting ready to release a new version soon.