CauldronDevelopmentLLC/CAMotics #139: Can Browse to Network Drives in Windows

ajayre, 2015-05-18:
All my CNC files are on a network share, mapped as a drive in Windows. OpenSCAM doesn't display or let me choose anything but the C: drive. This is a fundamental problem.

jcoffland, 2015-05-18:
In Windows, if I click on ```My Computer``` I can see other drivers.  Can you show me a screenshot of the problem?  Thanks for all the feedback.

ajayre, 2015-05-18:
I've uninstalled the application so I can't make any screenshots, sorry. On my PC I have mapped network drives M: and P:. Going to My Computer in OpenSCAM only showed me C:. I am using Win7 64-bit.

jcoffland, 2015-05-18:
Ok, I don't have a network drive to test this with.  There may be a limitation on which drives it can access.  I don't know why though.

lome64, 2016-05-26:
Here's a screenshot of the problem.
Windows7 64bit CAMotics 1.0.6
[image]

jcoffland, 2016-05-26:
As a workaround you should be able to type in ``\\server\path`` in to the location bar at the top.

closed 2016-05-26 by jcoffland

lome64, 2016-05-26:
>> As a workaround you should be able to type in \\server\path in to the location bar at the top.

Sorry. The path in that box cannot be edited. It's just a selection field.

  Lothar

ajayre, 2016-05-26:
Entering obscure paths into boxes isn't very user-friendly. My advice would
be to only mark this as closed when a user-friendly solution has been
implemented. :)

jcoffland, 2016-05-26:
Ah, I didn't realized I had closed the issue.  I believe I've fixed this in the source code.  It should be corrected in the next release.

commit 1ae1e2d0 (2016-05-26): Use native file dialogs to fix #139