CauldronDevelopmentLLC/CAMotics #140: Default Resolution

ajayre, 2015-05-18:
When starting the application for the first time the default units are mm and the default resolution is some strange value like 0.331262. This makes entering the size of the workpiece difficult. The default resolution should make sense for the default units, i.e. 0.1mm.

jcoffland, 2015-05-18:
The resolution you are referring to is automatically calculated based on the size of your tool path.  It's not actually related to the problems with editing the size of the workpiece.  But you are right there is a problem with editing the workpiece size.

jcoffland, 2015-08-23:
Aside from the confusion about what "resolution" means in the app, this is a duplicate of #149.

closed 2015-08-23 by jcoffland