CauldronDevelopmentLLC/CAMotics #141: Wrong Context Menu

ajayre, 2015-05-18:
Right-clicking on parts or workpieces in the both at the top left always shows the tools context menu. There should be context menus that match what is being right-clicked on.

jcoffland, 2015-05-18:
I'm not sure what you mean here.  There's only one context menu and it has options for adding and removing files or tools.

ajayre, 2015-05-18:
The items in the context menu should change depending on exactly what in the tree was right-clicked on. For me if I right-click on the workpieces area I only get an option to add tools.

jcoffland, 2015-05-18:
The same context menu comes up no matter where you click but I see what you mean but I think it does work fine how it is.  I think you probably just didn't see the options to add and remove files.  I'm quite positive they are there.

closed 2015-05-18 by jcoffland