CauldronDevelopmentLLC/CAMotics #143: Name

ajayre, 2015-05-18:
I don't mean to be rude. I offer this only as constructive feedback. When I first came across this application my first reaction was that the name was "suspicious". These days with the internet like it is "scam" is a word used frequently and doesn't mean anything good.

Now I understand "CAM" is what is going on here, but I think it is unfortunate that you have ended up with "SCAM" in the name.

So my first thought was "can I trust this application I have never heard of that has scam in the name"?

As you can see I have installed and tried it, so in the end it was fine for me, but it might not be for others. So I suggest you might want to look at this and tweak it a bit. Why not OpenCAMS or some other kind of rearrangement?

Best wishes for your project. I hope to see it mature over the coming years.

jcoffland, 2015-05-18:
It was a joke but I'm planning on changing the name to CAMotics.

jcoffland, 2015-08-23:
It has now been renamed.

closed 2015-08-23 by jcoffland