CauldronDevelopmentLLC/CAMotics #148: OpenSCAM fails with QT assertion error if qt is compiled in debug mode

wrtlprnft, 2015-07-16:
My system QT is compiled in debug mode, so it checks assertions. When starting ./openscam, before any GUI shows up, I get the following error, followed by program termination:

    ASSERT failure in QAbstractItemView::setModel: "The parent of a top level index should be invalid", file /var/tmp/portage/dev-qt/qtgui-4.8.7/work/qt-everywhere-opensource-src-4.8.7/src/gui/itemviews/qabstractitemview.cpp, line 670

Using gdb and a breakpoint in qt_assert_x, I get the following backtrace:

    #0  0x00007ffff2da2b00 in qt_assert_x(char const*, char const*, char const*, int) () from /usr/lib64/qt4/libQtCore.so.4
    #1  0x00007ffff39a1b8d in QAbstractItemView::setModel(QAbstractItemModel*) () from /usr/lib64/qt4/libQtGui.so.4
    #2  0x00007ffff39be380 in QHeaderView::setModel(QAbstractItemModel*) () from /usr/lib64/qt4/libQtGui.so.4
    #3  0x00007ffff39f6c89 in QTreeView::setModel(QAbstractItemModel*) () from /usr/lib64/qt4/libQtGui.so.4
    #4  0x000000000042f753 in OpenSCAM::QtWin::loadProject() ()
    #5  0x0000000000436991 in OpenSCAM::QtWin::newProject() ()
    #6  0x000000000042a535 in OpenSCAM::QtApp::run() ()
    #7  0x000000000042765a in _start ()

The assertions within QT are the follwing (the latter line is the one that's failing):

    // These asserts do basic sanity checking of the model
    Q_ASSERT_X(d->model->index(0,0) == d->model->index(0,0),
               "QAbstractItemView::setModel",
               "A model should return the exact same index "
               "(including its internal id/pointer) when asked for it twice in a row.");
    Q_ASSERT_X(!d->model->index(0,0).parent().isValid(),
               "QAbstractItemView::setModel",
               "The parent of a top level index should be invalid");

When I override the assertion by telling gdb to directly return from qt_assert_x, the program appears to run fine. I'm using QT version 4.8.7 and OpenSCAM rev 3790daa.

jcoffland, 2015-07-16:
Thanks, I think I've fixed it in the ``dev`` branch.

jcoffland, 2015-07-16:
Copied the fix to ``master``.

closed 2015-07-16 by jcoffland

wrtlprnft, 2015-07-16:
thanks for the quick fix!