CauldronDevelopmentLLC/CAMotics #150: OpenSCAM build in Windows 32 from Source

kelly17c, 2015-08-04:
I know it's been said that building OpenSCAM in Windows is difficult, but I'm trying to do it. I want to be able to edit the program and configure it to work with other programs. I'm having troubles.

I believe I've followed the Building_in_Windows instructions:
- I have a env.bat file (which is similar to the one on Building_in_Windows page). However, it seems as though the names as variables are different (QTDIR vs QT4DIR) and/or some are missing (BOOST_VERSION)?
- I'm using the new VS 2015
- Git is installed and available on the CL
- I'm using Python 2.7
- SCons 2.3.5 is installed to Python27/Scripts
- I've tried both Boost 1.40 and 1.58. I'm not 100% confident the env.bat has the correct location for this as when I unzip the folder, it's kind of ambiguous which directory to link the variable to.
- I've attempted to download V8: `depot_tools clone`, `gclient`, `fetch v8`, `cd v8`, `git checkout 3.14.5.6`, and `python build\gyp_v8` were (apparently) successful. However, when I run `devenv /build Release build\All.sln` the final line of output looks like `Build: 5 succeeded, 11 failed, 0 up-to-date, 0 skipped`. Is this expected?
- I continue on to build C!: `git clone [url]` and `cd cbang` are successful. Running `scons` seems to fail (even when I append `strict=0`). Here is the output. From what I understand, this indicates it is not referencing C! but I have no idea how to fix this.

```
C:\Users\Documents\CAM-SOFTWARES\openSCAM\Build\cbang>scons strict=0
scons: Reading SConscript files ...
  Compiler: cl (msvc)
  Platform: win32
  Mode: msvc
running with -j 1
Checking for C header file zlib.h... no
Need header zlib.h(cached) error: no result
Checking for C header file bzlib.h... no
Need header bzlib.h(cached) error: no result
Checking for C header file expat.h... no
Need C header expat.h(cached) error: no result
Checking for C library wsock32... yes
Checking for C library sqlite3... no
Need library sqlite3(cached) error: no result
Checking for C header file event.h... yes
Checking for C library event... no
Need library event(cached) error: no result
Checking for C library leveldb... no
Checking for C header file mysql/mysql.h... no
Checking for boost version >= 1.40...no
Missing boost version 1.40(cached) error: no result
Checking for C header file openssl/ssl.h... no
Checking for C library libeay32... no
Need openssl(cached) error: no result
Checking for C library winmm... yes
Checking for C++ header file v8.h... no
Need C++ header v8.h(cached) error: no result
Checking for C library wsock32... yes
Checking for C library setupapi... yes
Checking for C header file valgrind/valgrind.h... no
Checking for C header file valgrind/drd.h... no
NameError: name 'compiler_mode' is not defined:
  File "C:\Users\Documents\CAM-SOFTWARES\openSCAM\Build\cbang\SConstruct
", line 59:
    variant_dir = 'build/' + lib))
  File "C:\Python27\Scripts\..\Lib\site-packages\scons-2.3.5\SCons\Script\SConsc
ript.py", line 614:
    return method(*args, **kw)
  File "C:\Python27\Scripts\..\Lib\site-packages\scons-2.3.5\SCons\Script\SConsc
ript.py", line 551:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Python27\Scripts\..\Lib\site-packages\scons-2.3.5\SCons\Script\SConsc
ript.py", line 260:
    exec _file_ in call_stack[-1].globals
  File "C:\Users\Documents\CAM-SOFTWARES\openSCAM\Build\cbang\build\expa
t\SConscript", line 14:
    if compiler_mode == 'gnu': env.Append(CFLAGS = ['-fexceptions'])

C:\Users\Documents\CAM-SOFTWARES\openSCAM\Build\cbang>
```

I continue on regardless, because I believe this should work, I've done everything right...right?
- I download the .exe version of Freetype 2.3.5. GnuWin32 is installed to my program files, so I link `Freetype2_Home` here.
- I have Qt 4.5.3 installed
- As suggested, I installed the whole GTK+ package (version 3.6.4)

- Now I attempt to run `git clone [OpenSCAM Url]`, `cd openscam`, `git checkout dev`, `scons` after setting the environment. I know something isn't right:

```
C:\Users\Documents\CAM-SOFTWARES\openSCAM>build\env.bat

C:\Users\Documents\CAM-SOFTWARES\openSCAM>git clone https://github.com/c
auldrondevelopmentllc/openscam.git
Cloning into 'openscam'...
remote: Counting objects: 2494, done.
remote: Total 2494 (delta 0), reused 0 (delta 0), pack-reused 2494R
Receiving objects: 100% (2494/2494), 9.06 MiB | 7.26 MiB/s, done.
Resolving deltas: 100% (1473/1473), done.
Checking connectivity... done.

C:\Users\Documents\CAM-SOFTWARES\openSCAM>cd openscam

C:\Users\Documents\CAM-SOFTWARES\openSCAM\openscam>git checkout dev
Branch dev set up to track remote branch dev from origin.
Switched to a new branch 'dev'

C:\Users\Documents\CAM-SOFTWARES\openSCAM\openscam>scons
scons: Reading SConscript files ...
  Compiler: cl (msvc)
  Platform: win32
  Mode: msvc
running with -j 1
Checking for C header file zlib.h... no
Exception: Need header zlib.h:
  File "C:\Users\Documents\CAM-SOFTWARES\openSCAM\openscam\SConstruct",
line 48:
    conf.CBConfig('cbang')
  File "C:\Python27\Scripts\..\Lib\site-packages\scons-2.3.5\SCons\SConf.py", li
ne 661:
    ret = self.test(context, *args, **kw)
  File "C:\Users\Documents\CAM-SOFTWARES\openSCAM\Build\cbang\config\__i
nit__.py", line 166:
    ret = env.cb_methods[name](conf, **kwargs)
  File "C:\Users\Documents\CAM-SOFTWARES\openSCAM\Build\cbang\config\cba
ng\__init__.py", line 97:
    conf.CBConfig('cbang-deps', local = False)
  File "C:\Python27\Scripts\..\Lib\site-packages\scons-2.3.5\SCons\SConf.py", li
ne 661:
    ret = self.test(context, *args, **kw)
  File "C:\Users\Documents\CAM-SOFTWARES\openSCAM\Build\cbang\config\__i
nit__.py", line 166:
    ret = env.cb_methods[name](conf, **kwargs)
  File "C:\Users\Documents\CAM-SOFTWARES\openSCAM\Build\cbang\config\cba
ng\__init__.py", line 32:
    conf.CBConfig('zlib', not local)
  File "C:\Python27\Scripts\..\Lib\site-packages\scons-2.3.5\SCons\SConf.py", li
ne 661:
    ret = self.test(context, *args, **kw)
  File "C:\Users\Documents\CAM-SOFTWARES\openSCAM\Build\cbang\config\__i
nit__.py", line 166:
    ret = env.cb_methods[name](conf, **kwargs)
  File "C:\Users\Documents\CAM-SOFTWARES\openSCAM\Build\cbang\config\zli
b\__init__.py", line 7:
    conf.CBRequireHeader('zlib.h')
  File "C:\Python27\Scripts\..\Lib\site-packages\scons-2.3.5\SCons\SConf.py", li
ne 661:
    ret = self.test(context, *args, **kw)
  File "C:\Users\Documents\CAM-SOFTWARES\openSCAM\Build\cbang\config\__i
nit__.py", line 122:
    raise Exception, 'Need header ' + hdr

C:\Users\Documents\CAM-SOFTWARES\openSCAM\openscam>
```

This is basically where I'm stuck. I think I've checked all the dependencies. Also, for reference here is my env.bat file (I know there are some redundancies, it was to try and cover all my bases):

```
@echo off

set BUILD_ROOT=C:\Users\Documents\CAM-SOFTWARES\openSCAM\Build
set TARGET_ARCH=x86

call "C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\vcvarsall.bat" x86

set PATH=%PATH%;%BUILD_ROOT%\depot_tools;C:\Python27\Scripts;C:\Program Files (x86)\NSIS

set BOOST_SOURCE=%BUILD_ROOT%\boost_1_40_0
Set BOOST_VERSION=1.40

set OPENSSL_HOME=%Build_Root%\openssl-1.0.2d

set V8_HOME=%BUILD_ROOT%\depot_tools\v8
set CBANG_HOME=%BUILD_ROOT%\cbang
set QTDIR=%BUILD_ROOT%\Qt4.5.3
set CAIRO_HOME=%BUILD_ROOT%\gtk\include\cairo
set FREETYPE2_HOME=C:\Program Files (x86)\GnuWin32
set GTK_HOME= %BUILD_ROOT%\gtk
Set GT4DIR=%BUILD_ROOT%\4.5.3
```

- Also, I've chanced upon this document/url: `https://cauldrondevelopment.com/svn/openscam/trunk/buildbot/slaves/Windows-XP-32bit/env.bat` which makes it seem as though the env.bat file should be more complex than what I have...
- And, I've seen that openSSL is needed for the Windows build, but it's not mentioned on the main info page. Is it needed?

jcoffland, 2015-08-04:
First off you cannot continue with out successfully building cbang.  I just checked in a change which should fix the ``NameError: name 'compiler_mode' is not defined`` problem.  So please update your cbang repo.

However, you have another bigger problem with your cbang build.  You are missing libv8.  See the line ``Checking for C++ header file v8.h... no``?  You can build cbang with out libv8 but OpenSCAM needs it.

kelly17c, 2015-08-04:
I did see mentions of libv8, however it is not mentioned in Building_in_Windows: only V8 is mentioned there, and as I mentioned there were issues with that build as well. I am attempting to download a different version of VS as it may not be compatible... I'm not sure, this is all quite new (and advanced) for me.
I will try using VS2013 and then ensure that libv8 is downloaded. Hopefully there is some progress. I'll update when that is done.

I've updated C!, thanks for that.
Also thanks for your help. I'd really like to see this work in Windows for me!

jcoffland, 2015-08-04:
libv8 is the library provided by v8.  Good luck.

kelly17c, 2015-08-05:
I'm having absolutely zero luck in download v8. The output I continually get is: 

```
========== Build: 0 succeeded, 11 failed, 5 up-to-date, 0 skipped ==========
```

Essentially, platform/mutex.h cannot be found (that's what is causing the errors). I cannot find it anywhere either.
From what I understand, using VS2013 might be my issue... can you offer any help with this? Or anything at all...?

jcoffland, 2015-08-05:
Building v8 is difficult.  I don't have any instructions for this other than what's on the Google Website.  Also, the only version I know works with OpenSCAM is 3.14.5.8, which is now quite old.  I plan to upgrade the interface sometime soon.  You might search around and see if you can find any prebuilt binaries.

kelly17c, 2015-08-07:
For now, I've had to quit trying to build from source. V8 proved beyond my abilities to compile. When there is an upgrade I would love to try to use OpenSCAM from source again. In the meantime, I will be using the binary.

kelly17c, 2015-08-07:
Apparently I haven't yet completely given up... I want this to work.
In this command `python build\gyp_v8`, what type of file should gyp_v8 be?

jcoffland, 2015-08-07:
It's a Python script.

closed 2023-08-10 by jcoffland