CauldronDevelopmentLLC/CAMotics #157: Naming conflict among libraries

DougCoffland, 2015-10-03:
If a calling program calls a function in a library and that same calling program accesses functions within another library and that other library also accesses functions within the first library an error is generated.  Here's a screenshot.

[image]

jcoffland, 2015-10-03:
What's in the files?

DougCoffland, 2015-10-03:
Did you get my e-mail?  I wasn't able to attach the files to the issue.

Doug

jcoffland, 2015-10-05:
I did.  It's possible to insert them like this:

test.tpl:
```
console.log("starting");
var t = require("./test.js");
var t2 = require("./test2.js");
t2.test2();
t.test();
```

test.js:
```
/******************************************************************************\

    TextLibrary.js is an Open-Source CAM software.
    Copyright (C) 2011-2015 Buildbotics LLC

    This program is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 2 of the License, or
    (at your option) any later version.

    This program is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with this program.  If not, see <http://www.gnu.org/licenses/>.

\******************************************************************************/
var t2 = require('./test2.js');
(function ()
{
	console.log("reading text library");
	
	var T = {};
	module.exports = T;
		
	T.test = function() {
		console.log('hello from test1.js');
		t.test2();
	};
	
})();
```

test2.js:
```
(function ()
{
	console.log("reading text library");
	
	var T = {};
	module.exports = T;
		
	T.test2 = function() {
		console.log('hello from test2.js');
	};
	
})();
```

jcoffland, 2015-10-05:
The problem is that ``t`` is not defined at ``test.js`` line 30.  That's to be expected.  I think you are confusing this with ``t`` not being defined at ``test.tpl`` line 5.

I get this output:

```
G21
INFO(1):starting
INFO(1):reading text library
INFO(1):reading text library
INFO(1):hello from test2.js
INFO(1):hello from test1.js
ERROR:Exception: ReferenceError: t is not defined
ERROR:    at Object.T.test (/mnt/mirror/home/jcoffland/projects/camotics/people/dcoffland/20151004/test.js:30:9)
ERROR:    at test.tpl:5:3
```

closed 2015-10-05 by jcoffland

jcoffland, 2015-10-05:
There was an issue with circular module dependencies.  E.g. if module ``a`` requires module ``b`` and module ``b`` requires module ``a``.  One of them would end up with a reference to an empty module.  I suspect  this may be what you were trying to demonstrate here.  This bug is now fixed but you'd have to rebuild from source or wait until I make the next release.

Here's an example which triggers that bug:

test.tpl:
```
require('./test1.tpl').foo();
require('./test2.tpl').foo();
```

test1.tpl:
```
var test2 = require('./test2.tpl');

module.exports = {
  foo: function () {
    print("foo 1\n");
    test2.bar();
  },

  bar: function () {
    print("bar 1\n");
  }
}
```

test2.tpl:
```
var test1 = require('./test1.tpl');

module.exports = {
  foo: function () {
    print("foo 2\n");
    test1.bar();
  },

  bar: function () {
    print("bar 2\n");
  }
}
```