CauldronDevelopmentLLC/CAMotics #158: Issue reading file

andrewhodel, 2015-10-06:
For some reason Camotics is prepending a character to a gcode file which does not exist when the file is viewed in any text editor.

Here is the file, http://millcrum.com/camotics_reads_wrong.gcode

This is in Camotics 1.0.1, binary downloaded today for OS X 10.10.2

The only way to fix it is to open the file in Camotics and edit it, on the first character this is prepended:

ï»¿; TOOL OPTIONS

You can remove the strange characters up to the ; and close the file to save it, then reopen it and it will work fine.

The strange part is that no other text editor, gcode interpreter, or program shows this character.

The file was generated as an HTML5 blob with this code - https://github.com/eligrey/FileSaver.js/blob/master/FileSaver.js

jcoffland, 2015-10-06:
Your file starts with a [UTF-8 byte order mark](https://en.wikipedia.org/wiki/Byte_order_mark).  From Wikipedia:

> The UTF-8 representation of the BOM is the byte sequence 0xEF,0xBB,0xBF. A text editor or web browser interpreting the text as ISO-8859-1 or CP1252 will display the characters ï»¿ for this.

It goes on to say:

> The Unicode Standard permits the BOM in UTF-8, but does not require or recommend its use. Byte order has no meaning in UTF-8

Note that [FileSaver.js prepends the BOM on this line](https://github.com/eligrey/FileSaver.js/blob/master/FileSaver.js#L76).

CAMotics follows LinuxCNC's GCode implementation which in turns follows the [RS274NGC specification](http://www.nist.gov/customcf/get_pdf.cfm?pub_id=823374).  Neither support Unicode or UTF-8 input.  Some GCode interpreters might allow UTF-8 but it's certainly not in the spec.  So technically your program is outputting invalid GCode.

All that said it would be nice if CAMotics understood UTF-8.

andrewhodel, 2015-10-06:
Ok that's fixed on Millcrum's side - https://github.com/andrewhodel/millcrum/commit/7524cefd5e817c553b8706cefd9b402138f9ddfb

With that done, it seems reasonable to test if the file has a UTF-8 byte order mark and open it accordingly if it does in CAMotics.

commit 3f36ebd7 (2015-10-06): Handle UTF-8 BOM.  Closes #158
closed 2015-10-06 by jcoffland