CauldronDevelopmentLLC/CAMotics #159: Tool path is not shown

perivar, 2015-10-08:
I have exactly the samme issue as the one described by Andy Dochterman in the google groups except I have Windows 8.1 Enterprise 64 bit OS:
"I recently installed openscam on a windows 7 32bit machine and the simulation works, but even with the 'show toolpath' on, I don't see any tool paths?  Anyone had this issue before?  On an older computer, I didn't have any issues..."

It seems everything works except that no tool paths or cut paths are shown (the red and green lines).
[image]

jcoffland, 2015-10-10:
It must be an issue with the Windows OpenGL driver.  Have you tried upgrading your graphics driver?  I have a Windows 7 install which I've tested this on but it does not have this problem.

yaroslavVl, 2015-10-17:
v 1.0.3 let's go!: win8.1 x64 (intel HD Graphics 4000, NVIDIA GeForce GT 630M) - no tool path;
win10 x64 (Mobile intel 4 series express chipset family (WDDM1.1)) - no problem;
win7 x64 (AMD Radeon HD 8670D) - no problem;
win8.1 x64 (integrated Intel HD graphics, Nvidia GT640M (ASUS All-in-one PC ET2701INTI)) - no problem;
win10 x64 (Mobile intel 965 express chipset family (WDDM1.1)) - no problem.

jcoffland, 2015-10-17:
Thanks for the report.  I have a couple of questions.  Are there any errors in the console?  You can open the console by selecting ``View`` -> ``Show Console`` in CAMotics.  Errors should be in red and warnings in yellow.  What NVidia driver versions do you have on the win8.1 machines?

yaroslavVl, 2015-10-17:
console (there is only white text):
Opening C:/Program Files (x86)/CAMotics/examples/camotics/camotics.xml
Tool Path bounds ((0,0,-1.5), (119.349,16.163,3)) dimensions (119.349,16.163,4.5)
Computing surface bounded by ((-7.87072,-2.72155,-4.35954), (127.369,19.0237,0.68454)) at 0.342275 grid resolution
Time: 2.26 secs Triangles: 141612 Triangles/sec: 62708.87 Resolution: 0.342275

yaroslavVl, 2015-10-17:
[image]

jcoffland, 2015-10-17:
How about the driver version for the GT640M?

yaroslavVl, 2015-10-17:
don't know - it's my work computer. i'm at home now.

jcoffland, 2015-10-17:
I'm wondering if your work computer has a newer version of the graphics driver.  Could you try upgrading your graphics driver on the GT 630M?  I'm curious to know if that fixes it.

It looks like Toshiba has newer drivers: Version: 9.18.13.5328 http://support.toshiba.com/support/viewContentDetail?contentId=4007998

yaroslavVl, 2015-10-17:
ok. i'll try. but maybe downloading will take about 1 hour

perivar, 2015-10-17:
Hi,

I can add that the PC I have tested with is running Windows 8.1. Enterprise
x64 with Intel HD Graphics 4000.
And as yaroslav wrote, no toolpath is showing. I am running version 1.0.1.
Unfortunately I haven't had time to test CAMotics on other PCs yet.

Cheers,
Per Ivar

perivar, 2015-10-17:
Another note from me:
When I try to update the driver sofware from the device manager Windows
claims I am using the latest version, which is Intel Corp,
version 10.18.10.3621, driver date 17/05/2014.
(Intel(R) HD Graphics 4000)

/Per Ivar

jcoffland, 2015-10-17:
That's interesting.  Both machines which are failing have an Intel HD Graphics 4000 card.  It must not be the NVidia driver.

jcoffland, 2015-10-17:
@perivar Are you able to update your graphics driver with the [Intel driver update utility](http://www.intel.com/p/en_US/support/detect?iid=dc_iduu)?

perivar, 2015-10-17:
Hi,
Thanks for looking into this.
I will give it a shot tomorrow morning. In my timezone it's getting late now (01:10).

Regards,
Per Ivar

jcoffland, 2015-10-17:
Ok, thanks!

perivar, 2015-10-18:
Hi,
Unfortunately the Intel Update failed on the graphic hardware with the
following message:
"Graphics Drivers
Error Message: Computer Manufacturer Graphics Driver Detected.
Error: A customized computer manufacturer driver is installed on your
computer. The Intel® Driver Update Utility is not able to update the
driver. Installing a generic Intel® driver instead of the customized
computer manufacturer driver can cause technical issues. Contact your
computer manufacturer for the latest driver for your computer."

So still no luck, but I will try if I can find another driver that might
work better.

Regards,
Per Ivar

perivar, 2015-10-18:
New update:
I have not found any way to upgrade the graphic card drivers since Windows
thinks it's the latest one and using the latest driver from the Intel site
fails (the installer stops because the driver is not approved by HP - my
laptop is a HP Elitebook 2570p).

The same ouput at the console as yaroslav described.
It might seem like the issue is bad OpenGL support in the HP Intel HD 4000
driver.
https://communities.intel.com/message/228133

So still no luck.

Regards,
Per Ivar

yaroslavVl, 2015-10-18:
This version of the graphics driver is not compatible with this version of the operating system((((

jcoffland, 2015-10-19:
I think it is a problem with the Intel HD4000 OpenGL implementation.  I've made some modifications to the code that might help.  It will be in the next version.  Thanks for the help.

yaroslavVl, 2015-10-19:
win7 x64 intel HD Graphics Family 8.15.10.23.21 (06.03.2011) VEN_8086&DEV_0102&CC_0300 - no problem

jcoffland, 2015-10-27:
Does 1.0.4 solve this problem?

perivar, 2015-10-27:
Unfortunately version 1.0.4 still did not fix the issue with Intel HD Graphics 4000 on Windows 8.1. Enterprise x64. I understand from yaroslav that he got the same Intel HD graphic card (4000 ?) to work using win7 but as this is a company laptop I don't have that option.
But keep up the good work Joseph!

perivar, 2015-10-27:
Sorry, I closed the issue by accident...

yaroslavVl, 2015-10-27:
no(((
[image]

jcoffland, 2015-10-28:
I've created a new release which you can find below.

  http://camotics.org/releases/public/release/camotics/windows-xp-32bit/v1.0/camotics_1.0.5_x86.exe

In this release there are some new settings under ``Edit`` -> ``Settings`` ->``Advanced``.  Here you can disable path virtual buffer objects.  Please let me know if disabling VBOs solves this problem.

perivar, 2015-10-28:
Im sorry but it still doen't work.

[image: Inline image 2]

Thanks for trying.
I don't know if this is related at all, but I have tried out another
software for something completely different lately and the normal OpenGL4
version of that software crashes on my machine, but the OpenGL3.3 version
works.
(I am referring to a texture generation tool called AwesomeBump).
This is some log output from that program (I just start it up and read the
top of the log file):
[28/10/2015 08:01:07] {Debug} Checking OpenGL version...
[28/10/2015 08:01:07] {Debug} Widget OpenGL:  4 . 0
[28/10/2015 08:01:07] {Debug} Context valid:  true
[28/10/2015 08:01:07] {Debug} OpenGL information:
[28/10/2015 08:01:07] {Debug} VENDOR:  Intel
[28/10/2015 08:01:07] {Debug} RENDERER:  Intel(R) HD Graphics 4000
[28/10/2015 08:01:07] {Debug} VERSION:  4.0.0 - Build 10.18.10.3621
[28/10/2015 08:01:07] {Debug} GLSL VERSION:  4.00 - Build 10.18.10.3621
[28/10/2015 08:01:07] {Debug} Version: 4

/Per Ivar

jcoffland, 2015-10-28:
So you got the same results both with VBOs enabled and disabled?  Did you reload the simulation after changing this setting?

yaroslavVl, 2015-10-28:
[image]

yaroslavVl, 2015-10-28:
still not (disable, save; disable reload save)

perivar, 2015-10-28:
Yes I did reload the simulation after changing the settings. In fact I tried all the combinations one after one. Saved one combination and restarted the application (Quit and then start again). I could not see any differences between any of the 4 combinations.
Just a suggestion that could make debugging easier: What if you add some sort of a log.file that dump data (possibly focusing on the OpenGL stuff) on application startup. Then we could send you those for the machines that fail.

jcoffland, 2015-10-28:
Damn it!.  I'm noticing also that your bounding box is missing too.  Looks like it's drawing solids fine but not lines.  I've got some more ideas if you guys are willing to keep trying them out.

perivar, 2015-10-28:
I think you are right, all the lines as missing.
Since I do want this to work on my machine I will happily help. Thanks for
creating such a tool!
/Per Ivar

yaroslavVl, 2015-10-28:
Ok!!! I will be glad to continue

jcoffland, 2015-10-28:
Ok, here's a new version that I'm hopeful will fix the problem.

  http://camotics.org/releases/public/release/camotics/windows-xp-32bit/v1.0/camotics_1.0.6_x86.exe

I found that I was unnecessarily enabling a deprecated option in OpenGL which affects line drawing.  I removed this option because it wasn't doing anything anyway.

As an added bonus this version includes a 4.5x speed up for sufficiently large simulations.

perivar, 2015-10-28:
Well done, it works!!
I will test it more out tomorrow but I think you have sorted it all out now.

[image: Inline image 1]

Cheers,
Per Ivar

jcoffland, 2015-10-28:
That's great news!

yaroslavVl, 2015-10-29:
[image]
Great!

jcoffland, 2016-09-24:
This was fixed in v1.0.6.

closed 2016-09-24 by jcoffland