CauldronDevelopmentLLC/CAMotics #161: Position slider doesn't work

kg5000, 2015-10-21:
Using v1.03 for Windows (but happens in Debian Whezzy, too). With some G-Code files the position slider will not move. The file will load and render but the position slider will remain on the left side. When moved it will immediately jump back to the left side. One G-Code file that causes the problem is 39,000 lines long, approximately 940 KB.

I can provide a G-Code file example if needed. I could not attach to this issue report.

jcoffland, 2015-10-21:
I've fixed this already in the upcoming version.  The problem is that when a GCode file does a rapid move before setting the feed rate CAMotics computes the time wrong.  This causes the slider bar to not move because it thinks the simulation takes zero time.

I will be making a new release soon.  In the meantime you can work around this by adding a default feed rate like ``F10`` near the top of your GCode file.

Thanks for the report.

kg5000, 2015-10-22:
Thanks for the quick response Joseph. I added a feed speed to the first G0
instruction and it worked like a charm! I look forward to an updated
release soon. Many thanks for such a cool application

Ken

jcoffland, 2015-10-28:
Fixed in v1.0.4.

closed 2015-10-28 by jcoffland