CauldronDevelopmentLLC/CAMotics #162: Build failure

f3nix, 2015-10-25:
Hi,
I have a problem while trying to build the latest camotics on debian testing. Here is an error log:

```g++ -o camotics -Wl,-S -Wl,-x -pthread -Wl,-rpath=/usr/lib build/camotics.o build/qrc_camotics.o libCAMotics.a libclipper.a -L/home/f3nix/devel/camotics/cbang/lib -lcairo -lv8 -lGLU -lGL -lpng -lz -lfreetype -lcbang -lv8 -lssl -lcrypto -lboost_regex -lboost_filesystem -lboost_system -lboost_iostreams -lre2 -levent -lsqlite3 -ldl -lexpat -lbz2 -lz -lpthread -lQt5OpenGL -lQt5Widgets -lQt5Gui -lQt5Core
libCAMotics.a(Simulation.o): In function `CAMotics::Simulation::computeHash[abi:cxx11]() const':
Simulation.cpp:(.text+0x616): undefined reference to `cb::Digest::Digest(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
Simulation.cpp:(.text+0x633): undefined reference to `cb::Digest::init(engine_st*)'
Simulation.cpp:(.text+0xc7a): undefined reference to `cb::Digest::toBase64[abi:cxx11](char, char, char)'
Simulation.cpp:(.text+0xe02): undefined reference to `cb::Digest::~Digest()'
Simulation.cpp:(.text+0x183e): undefined reference to `cb::Digest::~Digest()'
Simulation.cpp:(.text+0x1c72): undefined reference to `cb::Digest::~Digest()'
libCAMotics.a(Simulation.o): In function `cb::DigestStreamFilter::~DigestStreamFilter()':
Simulation.cpp:(.text._ZN2cb18DigestStreamFilterD2Ev[_ZN2cb18DigestStreamFilterD5Ev]+0xf): undefined reference to `cb::Digest::~Digest()'
libCAMotics.a(Simulation.o): In function `cb::DigestStreamFilter::~DigestStreamFilter()':
Simulation.cpp:(.text._ZN2cb18DigestStreamFilterD0Ev[_ZN2cb18DigestStreamFilterD5Ev]+0x13): undefined reference to `cb::Digest::~Digest()'
libCAMotics.a(Simulation.o):(.data.rel.ro._ZTIN2cb18DigestStreamFilterE[_ZTIN2cb18DigestStreamFilterE]+0x18): undefined reference to `typeinfo for cb::Digest'
libCAMotics.a(Simulation.o):(.data.rel.ro._ZTVN2cb18DigestStreamFilterE[_ZTVN2cb18DigestStreamFilterE]+0x20): undefined reference to `cb::Digest::size() const'
libCAMotics.a(Simulation.o):(.data.rel.ro._ZTVN2cb18DigestStreamFilterE[_ZTVN2cb18DigestStreamFilterE]+0x28): undefined reference to `cb::Digest::update(unsigned char const*, unsigned int)'
libCAMotics.a(Simulation.o):(.data.rel.ro._ZTVN2cb18DigestStreamFilterE[_ZTVN2cb18DigestStreamFilterE]+0x30): undefined reference to `cb::Digest::finalize()'
libCAMotics.a(Simulation.o):(.data.rel.ro._ZTVN2cb18DigestStreamFilterE[_ZTVN2cb18DigestStreamFilterE]+0x38): undefined reference to `cb::Digest::reset()'
collect2: error: ld returned 1 exit status
scons: *** [camotics] Error 1
scons: building terminated because of errors.```

Any idea what is going on?

Thanks!

Cheers,
Mateusz

jcoffland, 2015-10-25:
Looks like you may have built cbang with out OpenSSL.

f3nix, 2015-10-25:
Hi,

```Checking for openssl version >= 1.0.0...(cached) no                                                                                                                                                                                         
Insufficient OpenSSL version(cached) error: no result```

```f3nix@bigspider:~/devel/camotics$ dpkg -l | grep libssl-dev
ii  libssl-dev:amd64                                            1.0.2d-1                            amd64        Secure Sockets Layer toolkit - development files
f3nix@bigspider:~/devel/camotics$ ```

Hmm. Something weird is going on here. Will have to try with fresh sources.

Thanks for the tip.

Cheers,
Mateusz

jcoffland, 2015-10-25:
Since it says ``(cached)``, perhaps you installed OpenSSL after the first time you built cbang.  Often what happens is people build cbang, then try to build CAMotics.  CAMotics complains about not having OpenSSL so they install OpenSSL, rebuild cbang and CAMotics and run in to the error your seeing because cbang had cached it's detection of OpenSSL and didn't rebuild.  Building with fresh sources would work but you can also just do this in cbang:

```
cd cbang
rm -rf .scon*
scons
```
This deletes cbang's database and configuration test files.

f3nix, 2015-10-25:
Hi,
thank You! That worked. I've managed to compile and create a package for Debian. Problem solved.

Thanks for Your great work! :)

Cheers,
Mateusz

jcoffland, 2015-10-25:
Glad I could help.

closed 2015-10-25 by jcoffland