CauldronDevelopmentLLC/CAMotics #167: file names including spaces are not correctly saved in the project file

ymargarin, 2015-12-02:
I just edited the xml file and replaced the spaces in the file names by %20 and opened again the project file. It worked.
Please fix, thanks

jcoffland, 2015-12-15:
I thought I fixed this. It was issue #50.  I'll look in to it. Thanks for the report.

ymargarin, 2015-12-15:
Hi,
Thanks for the feedback.

To be more factual, since I was able to load the files after substituting the space characters by %20, I thought that the issue was related to spaces. I saw the issue #50 and thought that the fix wasn't complete, hence my message.

After a more thorough review, I'm realizing that the issue might be linked to both the presence of " ! " and the space characters. My files have this structure:
!Motor Command Board_ver 2.2.drl.ngc ; !Motor Command Board_ver 2.2-B_Cu.ngc ; ...
and the complete path is:
"D:\My_Vault\!CAD\Kicad\My_Projects\Remote receiver for shade motor\!Motor Interface Unit\!Motor Command Board\Manufacturing\NEW_ Ver 2.2\!Motor Command Board_ver 2.2-B_Cu.ngc"
the " ! " is here to provide a hierarchy in explorer and I'm extensively using it.

The result after "save project as" and "load project" is:
[image]

I have a similar structure in another directory but the exclamation mark is in he path instead of being in the filename. Here's the complete path followed by a picture of the resulting list:
"D:\My_Vault\!CAD\Kicad\My_Projects\Remote receiver for shade motor\!Motor Interface Unit Design-2\Manufacturing\Timer board-Edge_Cuts.gbr.ngc".
[image].

I'm aware of some character avoidance within the CAD file filenames but until now, everything worked fine with kicad, mach3, flatcam, wings and a bunch of other programs.

Sorry for the misleading, I do hope that this clarification will help.
looking forward to ...
best regards

closed 2017-03-26 by jcoffland