CauldronDevelopmentLLC/CAMotics #168: Compilation trouble  / Fedora 22

B-Laurent, 2015-12-04:
Hello,

During compilation under linux fedora 22, I experienced some troubles around qt settings. My trouble are  similar as those reported here but this topic didn't solve the problem: 
https://github.com/CauldronDevelopmentLLC/CAMotics/issues/118

The trouble is around the qt_version setting in the SContruct file. In this file, if I change the following line 

# Qt5 tool
qt_version = env.get('qt_version')

by

# Qt5 tool
qt_version = os.environ.get('qt_version')

Everything goes well. I not able to write a patch, but I can help but I can do tests.

LAurent_B

jcoffland, 2015-12-14:
The way you are supposed to set ``qt_version`` is not as an environment variable but on the command line like this:

    scons qt_version=5

closed 2015-12-14 by jcoffland

B-Laurent, 2016-02-03:
Thanks for help
For fedora's users, here my script to compiled Camotics with qt4. 

[script_compileCAMotics-qt4.sh.txt](https://github.com/CauldronDevelopmentLLC/CAMotics/files/115394/script_compileCAMotics-qt4.sh.txt)

the same script with qt5 gives me a `Erreur de segmentation (core dumped)` message with the following command :

`./CAMotics/camotics`

jcoffland, 2016-02-14:
@B-Laurent unfortunately there seem to be a lot of bad builds of Qt on various operating systems.  On some platforms qt4 works better on others qt5.

B-Laurent, 2016-02-16:
If you need more informations about qt5 builds let me know.
I will keep you informed of the results of my future builds on fedora with qt4 and qt5.

srakitnican, 2017-07-09:
Just a note. If building with Qt5 for Fedora there is a build system requirement for command `lupdate-qt5` and `lrelease-qt5` that can be found on package `qt5-linguist`. Else, build system just silently falls back to use Qt4 when it can't find first command due to raised exception.

On a side note, a build with Qt5 for Fedora 25 works fine here.

jcoffland, 2017-07-09:
Should we add a dependency on ``qt5-linquist``?  Will that still work with Fedora 25?

srakitnican, 2017-07-09:
Yes, I've just added as a build dependency, it should be fine.