CauldronDevelopmentLLC/CAMotics #173: V-bit is simulated wrong?

vilemnovak, 2016-01-04:
Hello, 
here I compare results of V - carving with 90 degree V-bit in CAMotics and in Blender CAM. the 90 degree cutter was defined in CAMotics as width = 2*length.

It's quite obvious there is something wrong, but I am not sure what, seems like the V-bit angle is not perfectly correct, or the simulation happens somehow different from what I imagine.
In blenderCAM, the simulation is done by sampling a bitmap, so it has precision issues, but kind of works reliably.
The path comes from blenderCAM medial axis strategy, but I tested F-engrave and it was the same...

I am the main developer of Blender CAM by the way. - CAMotics is a great tool! Thank you for developing it.

[image]

[image]

jcoffland, 2016-01-04:
I noticed something similar recently.  I'm working on fixing it.  This problem occurs with slanting moves and conical bits.

Could you paste your GCode here or compare with the ``slant_test`` example so we have a common test case?  ``slant_test`` might be better since it's a very small tests for this particular issue.

vilemnovak, 2016-01-04:
adding my code -  BlenderCAM can only export gcode, so there's no way to compare for real...
[testvcarve.zip](https://github.com/CauldronDevelopmentLLC/CAMotics/files/77823/testvcarve.zip)

here is slant_test run here with 90 degrees cutter:
[image]

jcoffland, 2016-01-05:
I think I may have already fixed this in the latest code.  I've been working on a few different things so I'll have to take a closer look.  Thanks for the input.

closed 2017-03-26 by jcoffland