CauldronDevelopmentLLC/CAMotics #174: will not run on Ubuntu 15:10

StevenTyrer, 2016-01-06:
I have been using Openscam for sometime with great success, due to a software update I have had to re-install your software but cannot now get it to run.
I am using Ubuntu 15.10 64bit
Down loaded the 64bit Ubuntu version and installed with the software centre, if I try and run using the icon nothing happens, if I try running with the terminal I get the following

steven@steven-Vostro-1500:~$ camotics 
*********************************** CAMotics ***********************************
     Website: http://camotics.org/
   Copyright: (c) 2011-2015
      Author: Joseph Coffland <joseph@camotics.org>
Organization: Cauldron Development LLC
        Args: 
      Config: <none>
************************************ Build *************************************
     Version: 1.0.6
        Date: Oct 28 2015
        Time: 10:38:49
  Repository: Git
    Revision: db5e189b3ba5996a91635e8ddf1396ed2018c698
      Branch: master
    Compiler: GNU 4.2.1 Compatible Ubuntu Clang 3.0 (tags/RELEASE_30/final)
     Options: -std=gnu++98 -O3 -funroll-loops -msse2 -fPIC
    Platform: linux2 3.16.0-4-amd64
        Bits: 64
        Mode: Release
************************************ System ************************************
         CPU: Intel(R) Core(TM)2 Duo CPU T9300 @ 2.50GHz
      CPU ID: GenuineIntel Family 6 Model 23 Stepping 6
        CPUs: 2
      Memory: 3.85GiB
 Free Memory: 1.69GiB
     Threads: POSIX_THREADS
  OS Version: 4.2
 Has Battery: true
  On Battery: false
  UTC Offset: 0
         PID: 6018
         CWD: /home/steven
********************************************************************************
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x0
QMetaObject::connectSlotsByName: No matching signal for on_addTool()
QMetaObject::connectSlotsByName: No matching signal for on_editTool(uint)
QMetaObject::connectSlotsByName: No matching signal for on_removeTool(uint)
ERROR:Exception: SmartPointer: Can't dereference a NULL pointer!
steven@steven-Vostro-1500:~$

jcoffland, 2016-01-06:
Can you try the [debug version](http://camotics.org/releases/public/debug/camotics/ubuntu-precise-64bit/v1.0/camotics_1.0.6_amd64.deb) and post the error?

StevenTyrer, 2016-01-06:
Hi errors from running debug version, thanks for responding so quickly 

steven@steven-Vostro-1500:~$ camotics
*********************************** CAMotics ***********************************
     Website: http://camotics.org/
   Copyright: (c) 2011-2015
      Author: Joseph Coffland <joseph@camotics.org>
Organization: Cauldron Development LLC
        Args: 
      Config: <none>
************************************ Build *************************************
     Version: 1.0.6
        Date: Oct 28 2015
        Time: 10:38:49
  Repository: Git
    Revision: db5e189b3ba5996a91635e8ddf1396ed2018c698
      Branch: master
    Compiler: GNU 4.2.1 Compatible Ubuntu Clang 3.0 (tags/RELEASE_30/final)
     Options: -std=gnu++98 -Werror -fPIC
    Platform: linux2 3.16.0-4-amd64
        Bits: 64
        Mode: Debug
************************************ System ************************************
         CPU: Intel(R) Core(TM)2 Duo CPU T9300 @ 2.50GHz
      CPU ID: GenuineIntel Family 6 Model 23 Stepping 6
        CPUs: 2
      Memory: 3.85GiB
 Free Memory: 2.45GiB
     Threads: POSIX_THREADS
  OS Version: 4.2
 Has Battery: true
  On Battery: false
  UTC Offset: 0
         PID: 4281
         CWD: /home/steven
********************************************************************************
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x0
QMetaObject::connectSlotsByName: No matching signal for on_addTool()
QMetaObject::connectSlotsByName: No matching signal for on_editTool(uint)
QMetaObject::connectSlotsByName: No matching signal for on_removeTool(uint)
terminate called after throwing an instance of 'std::bad_cast'
  what():  std::bad_cast
Aborted (core dumped)
steven@steven-Vostro-1500:~$

jcoffland, 2016-01-06:
Strange that is a bit different error.  Could you trying something else:

    sudo apt-get install -y gdb
    gdb /usr/bin/camotics -c core

Run this in the same directory where CAMotics crashed and then type ``bt`` at the prompt.  This should give stack trace were it crashed and allow me to pin point the bug.

If this doesn't work It's possible the core file is named ``core.#####`` where ``#####`` is some number.  If you do an ``ls core.*`` you should be able to see the correct name and substitute it for ``core`` above.

Thanks!

StevenTyrer, 2016-01-06:
Hi, hope this helps
[sudo] password for steven: 
Sorry, try again.
[sudo] password for steven: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
gdb is already the newest version.
The following packages were automatically installed and are no longer required:
  linux-headers-4.2.0-22 linux-headers-4.2.0-22-generic
  linux-image-4.2.0-22-generic linux-image-extra-4.2.0-22-generic
Use 'apt-get autoremove' to remove them.
0 to upgrade, 0 to newly install, 0 to remove and 7 not to upgrade.
steven@steven-Vostro-1500:~$ gdb /usr/bin/camotics -c core
GNU gdb (Ubuntu 7.10-1ubuntu2) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/camotics...done.
[New LWP 4281]
[New LWP 4283]
[New LWP 4284]
[New LWP 4285]
[New LWP 4286]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `camotics'.
Program terminated with signal SIGABRT, Aborted.
#0  0x00007f0ba807e267 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:55
55	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
[Current thread is 1 (Thread 0x7f0ba9bf88c0 (LWP 4281))]
(gdb) bt
#0  0x00007f0ba807e267 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:55
#1  0x00007f0ba807feca in __GI_abort () at abort.c:89
#2  0x000000000067763d in __gnu_cxx::__verbose_terminate_handler() ()
#3  0x0000000000672066 in __cxxabiv1::__terminate(void (*)()) ()
#4  0x0000000000672093 in std::terminate() ()
#5  0x0000000000670f3e in __cxa_throw ()
#6  0x000000000066f652 in __cxa_bad_cast ()
#7  0x00000000006416aa in std::numpunct<char> const& std::use_facet<std::numpunct<char> >(std::locale const&) ()
#8  0x0000000000698d20 in cb::String::String(double) ()
#9  0x00000000006c0589 in cb::Option::setDefault(double) ()
#10 0x000000000053f77a in cb::SmartPointer<cb::Option, cb::DeallocNew<cb::Option>, cb::RefCounterImpl<cb::Option, cb::DeallocNew<cb::Option> > > cb::OptionMap::addTarget<double>(std::string const&, double&, std::string const&, char) ()
#11 0x000000000053aa60 in CAMotics::Project::Project(cb::Options&, std::string const&) ()
#12 0x000000000047f581 in CAMotics::QtWin::openProject(std::string const&) ()
#13 0x000000000047e936 in CAMotics::QtWin::loadDefaultExample() ()
#14 0x000000000047034d in CAMotics::QtApp::run() ()
#15 0x000000000046c8a7 in int cb::doApplication<CAMotics::QtApp>(int, char**)
    ()
#16 0x000000000046c7e2 in main ()
(gdb) Quit
(gdb)

jcoffland, 2016-01-06:
This looks like a locale issue.  It's having problems parsing a number from the project file at ``std::use_facetstd::numpunct<char >(std::locale const&)()``.  Which I'm guessing means you are in a country where you use ``,`` instead of ``.`` for the decimal place.

Try this:

    export LC_NUMERIC=C
    camotics

StevenTyrer, 2016-01-06:
Hi Joseph

Strange I am in Cardiff in the UK?

steven@steven-Vostro-1500:~$ export LC_NUMERIC=C
steven@steven-Vostro-1500:~$ camotics
*********************************** CAMotics ***********************************
     Website: http://camotics.org/
   Copyright: (c) 2011-2015
      Author: Joseph Coffland <joseph@camotics.org>
Organization: Cauldron Development LLC
        Args: 
      Config: <none>
************************************ Build *************************************
     Version: 1.0.6
        Date: Oct 28 2015
        Time: 10:38:49
  Repository: Git
    Revision: db5e189b3ba5996a91635e8ddf1396ed2018c698
      Branch: master
    Compiler: GNU 4.2.1 Compatible Ubuntu Clang 3.0 (tags/RELEASE_30/final)
     Options: -std=gnu++98 -Werror -fPIC
    Platform: linux2 3.16.0-4-amd64
        Bits: 64
        Mode: Debug
************************************ System ************************************
         CPU: Intel(R) Core(TM)2 Duo CPU T9300 @ 2.50GHz
      CPU ID: GenuineIntel Family 6 Model 23 Stepping 6
        CPUs: 2
      Memory: 3.85GiB
 Free Memory: 1.30GiB
     Threads: POSIX_THREADS
  OS Version: 4.2
 Has Battery: true
  On Battery: false
  UTC Offset: 0
         PID: 16022
         CWD: /home/steven
********************************************************************************
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x0
QMetaObject::connectSlotsByName: No matching signal for on_addTool()
QMetaObject::connectSlotsByName: No matching signal for on_editTool(uint)
QMetaObject::connectSlotsByName: No matching signal for on_removeTool(uint)
terminate called after throwing an instance of 'std::bad_cast'
  what():  std::bad_cast
Aborted (core dumped)

StevenTyrer, 2016-01-09:
Hi This is the results of your last recommendation

steven@steven-Vostro-1500:~$ export LC_NUMERIC=C
steven@steven-Vostro-1500:~$ camotics
*********************************** CAMotics ***********************************
     Website: http://camotics.org/
   Copyright: (c) 2011-2015
      Author: Joseph Coffland <joseph@camotics.org>
Organization: Cauldron Development LLC
        Args: 
      Config: <none>
************************************ Build *************************************
     Version: 1.0.6
        Date: Oct 28 2015
        Time: 10:38:49
  Repository: Git
    Revision: db5e189b3ba5996a91635e8ddf1396ed2018c698
      Branch: master
    Compiler: GNU 4.2.1 Compatible Ubuntu Clang 3.0 (tags/RELEASE_30/final)
     Options: -std=gnu++98 -Werror -fPIC
    Platform: linux2 3.16.0-4-amd64
        Bits: 64
        Mode: Debug
************************************ System ************************************
         CPU: Intel(R) Core(TM)2 Duo CPU T9300 @ 2.50GHz
      CPU ID: GenuineIntel Family 6 Model 23 Stepping 6
        CPUs: 2
      Memory: 3.85GiB
 Free Memory: 998.11MiB
     Threads: POSIX_THREADS
  OS Version: 4.2
 Has Battery: true
  On Battery: false
  UTC Offset: 0
         PID: 18774
         CWD: /home/steven
********************************************************************************
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x0
QMetaObject::connectSlotsByName: No matching signal for on_addTool()
QMetaObject::connectSlotsByName: No matching signal for on_editTool(uint)
QMetaObject::connectSlotsByName: No matching signal for on_removeTool(uint)
terminate called after throwing an instance of 'std::bad_cast'
  what():  std::bad_cast
Aborted (core dumped)

StevenTyrer, 2016-01-11:
Hi 
Un-installed   ver 1.0.6 and the installed ver 1.0.5 which work fine

jcoffland, 2016-01-11:
That's good to know.  I'll compare the code changes in these two versions to see if I can figure out what might have caused this regression.

iforce2d, 2016-01-20:
I'm not sure if this helps much, but I built from source on Fedora 20 with git clone from today and I'm having this issue too. The main window of the app shows for a brief moment and then closes with segmentation fault. I built everything again in debug mode, and this is the last part of valgrind output:
```
==8793== 
==8793== Conditional jump or move depends on uninitialised value(s)
==8793==    at 0x5ED57A: CAMotics::QtWin::event(QEvent*) (QtWin.cpp:1418)
==8793==    by 0x3B3A7CAE5B: QApplicationPrivate::notify_helper(QObject*, QEvent*) (in /usr/lib64/libQtGui.so.4.8.6)
==8793==    by 0x3B3A7D17B4: QApplication::notify(QObject*, QEvent*) (in /usr/lib64/libQtGui.so.4.8.6)
==8793==    by 0x6EC0CA: CAMotics::QApplication::notify(QObject*, QEvent*) (QApplication.cpp:34)
==8793==    by 0x3B3B5868FC: QCoreApplication::notifyInternal(QObject*, QEvent*) (in /usr/lib64/libQtCore.so.4.8.6)
==8793==    by 0x3B3A81E761: QWidget::setParent(QWidget*, QFlags<Qt::WindowType>) (in /usr/lib64/libQtGui.so.4.8.6)
==8793==    by 0x3B3A81EC09: QWidgetPrivate::init(QWidget*, QFlags<Qt::WindowType>) (in /usr/lib64/libQtGui.so.4.8.6)
==8793==    by 0x3B3A81F1E3: QWidget::QWidget(QWidgetPrivate&, QWidget*, QFlags<Qt::WindowType>) (in /usr/lib64/libQtGui.so.4.8.6)
==8793==    by 0x3B3AC92291: QDialog::QDialog(QDialogPrivate&, QWidget*, QFlags<Qt::WindowType>) (in /usr/lib64/libQtGui.so.4.8.6)
==8793==    by 0x3B3ACA36E9: QFileDialog::QFileDialog(QWidget*, QString const&, QString const&, QString const&) (in /usr/lib64/libQtGui.so.4.8.6)
==8793==    by 0x6DCD9B: CAMotics::FileDialog::FileDialog(CAMotics::QtWin&) (FileDialog.cpp:35)
==8793==    by 0x5DF281: CAMotics::QtWin::QtWin(cb::Application&) (QtWin.cpp:73)
==8793== 
==8793== Conditional jump or move depends on uninitialised value(s)
==8793==    at 0x38340572F5: ??? (in /usr/lib64/nvidia/libGL.so.331.89)
==8793== 
QMetaObject::connectSlotsByName: No matching signal for on_addTool()
QMetaObject::connectSlotsByName: No matching signal for on_editTool(uint)
QMetaObject::connectSlotsByName: No matching signal for on_removeTool(uint)
vex amd64->IR: unhandled instruction bytes: 0x4D 0x6F 0x74 0x69 0x63 0x73 0x31 0x32
vex amd64->IR:   REX=1 REX.W=1 REX.R=1 REX.X=0 REX.B=1
vex amd64->IR:   VEX=0 VEX.L=0 VEX.nVVVV=0x0 ESC=NONE
vex amd64->IR:   PFX.66=0 PFX.F2=0 PFX.F3=0
==8793== valgrind: Unrecognised instruction at address 0x867604.
==8793==    at 0x867604: ??? (in /wd1tb/chris/projects/gcodesim/CAMotics/camotics)
==8793==    by 0x3B3A847FA8: ??? (in /usr/lib64/libQtGui.so.4.8.6)
==8793==    by 0x3B3A848284: ??? (in /usr/lib64/libQtGui.so.4.8.6)
==8793==    by 0x383DA04C5C: _SmcProcessMessage (in /usr/lib64/libSM.so.6.0.1)
==8793==    by 0x383DE10BD6: IceProcessMessages (in /usr/lib64/libICE.so.6.3.0)
==8793==    by 0x3B3B59B379: QMetaObject::activate(QObject*, QMetaObject const*, int, void**) (in /usr/lib64/libQtCore.so.4.8.6)
==8793==    by 0x3B3B5E8A0D: QSocketNotifier::activated(int) (in /usr/lib64/libQtCore.so.4.8.6)
==8793==    by 0x3B3B5A458A: QSocketNotifier::event(QEvent*) (in /usr/lib64/libQtCore.so.4.8.6)
==8793==    by 0x3B3A7CAE5B: QApplicationPrivate::notify_helper(QObject*, QEvent*) (in /usr/lib64/libQtGui.so.4.8.6)
==8793==    by 0x3B3A7D17B4: QApplication::notify(QObject*, QEvent*) (in /usr/lib64/libQtGui.so.4.8.6)
==8793==    by 0x6EC0CA: CAMotics::QApplication::notify(QObject*, QEvent*) (QApplication.cpp:34)
==8793==    by 0x3B3B5868FC: QCoreApplication::notifyInternal(QObject*, QEvent*) (in /usr/lib64/libQtCore.so.4.8.6)
```

szpapas, 2016-02-21:
I build from source on 15.10 ubuntu. It works.

[image]

yaroslavVl, 2016-03-08:
ubuntu 15.10 (virtual machine)
camotics_1.0.6_amd64.deb
*********************************** CAMotics ***********************************
     Website: http://camotics.org/
   Copyright: (c) 2011-2015
      Author: Joseph Coffland <joseph@camotics.org>
Organization: Cauldron Development LLC
        Args: 
      Config: <none>
************************************ Build *************************************
     Version: 1.0.6
        Date: Oct 28 2015
        Time: 10:38:49
  Repository: Git
    Revision: db5e189b3ba5996a91635e8ddf1396ed2018c698
      Branch: master
    Compiler: GNU 4.2.1 Compatible Ubuntu Clang 3.0 (tags/RELEASE_30/final)
     Options: -std=gnu++98 -O3 -funroll-loops -msse2 -fPIC
    Platform: linux2 3.16.0-4-amd64
        Bits: 64
        Mode: Release
************************************ System ************************************
         CPU: Intel(R) Core(TM) i7-3630QM CPU @ 2.40GHz
      CPU ID: GenuineIntel Family 6 Model 58 Stepping 9
        CPUs: 1
      Memory: 1.45GiB
 Free Memory: 79.32MiB
     Threads: POSIX_THREADS
  OS Version: 4.2
 Has Battery: true
  On Battery: false
  UTC Offset: 3
         PID: 4144
         CWD: /home/virtualscience/Desktop/2
********************************************************************************
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x0
QMetaObject::connectSlotsByName: No matching signal for on_addTool()
QMetaObject::connectSlotsByName: No matching signal for on_editTool(uint)
QMetaObject::connectSlotsByName: No matching signal for on_removeTool(uint)
ERROR:Exception: SmartPointer: Can't dereference a NULL pointer!

yaroslavVl, 2016-03-08:
ubuntu 15.10 (virtual machine)
from sources - IT'S WORK!
[image]

*********************************** CAMotics ***********************************
     Website: http://camotics.org/
   Copyright: (c) 2011-2015
      Author: Joseph Coffland <joseph@camotics.org>
Organization: Cauldron Development LLC
        Args: 
      Config: <none>
************************************ Build *************************************
     Version: 1.0.7
        Date: Mar 8 2016
        Time: 15:02:17
  Repository: Git
    Revision: 084638a797d3200c1dd19c51ac33fabe45d157cb
      Branch: master
    Compiler: GNU 5.2.1 20151010
     Options: -std=gnu++98 -O3 -funroll-loops -ffast-math -mfpmath=sse
              -fno-unsafe-math-optimizations -msse2 -fPIC
    Platform: linux2 4.2.0-16-generic
        Bits: 64
        Mode: Release
************************************ System ************************************
         CPU: Intel(R) Core(TM) i7-3630QM CPU @ 2.40GHz
      CPU ID: GenuineIntel Family 6 Model 58 Stepping 9
        CPUs: 1
      Memory: 1.45GiB
 Free Memory: 378.29MiB
     Threads: POSIX_THREADS
  OS Version: 4.2
 Has Battery: true
  On Battery: false
  UTC Offset: 3
         PID: 27337
         CWD: /home/virtualscience/Desktop/2/CAMotics
********************************************************************************
Opening /usr/share/doc/camotics/examples/camotics/camotics.xml
Tool Path bounds ((0,0,-1), (119,16,3)) dimensions (119,16,4)
Computing surface bounded by ((59.5,7.5,-1.5), (59.5,7.5,-1.5)) at 0 grid resolution
Time: 0,00 secs Triangles: 0 Triangles/sec: 0,00

jcoffland, 2016-03-09:
Ubuntu continues to ship versions of the Qt libraries that are broken in different ways.  @yaroslavVl could you run the [debug version of v1.0.6](http://camotics.org/releases/public/debug/camotics/debian-testing-64bit/v1.0/camotics_1.0.6_amd64.deb) on Ubuntu 15.10 so we can see the full stacktrace of this error: ``ERROR:Exception: SmartPointer: Can't dereference a NULL pointer!``?

yaroslavVl, 2016-03-10:
hmmmm...
uninstall camotics 1.0.7, then install  debug version of v1.0.6 from your link.
it works.

(PS it works with ugly arcs #180 )

yaroslavVl, 2016-03-10:
*********************************** CAMotics ***********************************
     Website: http://camotics.org/
   Copyright: (c) 2011-2015
      Author: Joseph Coffland <joseph@camotics.org>
Organization: Cauldron Development LLC
        Args: 
      Config: <none>
************************************ Build *************************************
     Version: 1.0.6
        Date: Oct 28 2015
        Time: 10:39:06
  Repository: Git
    Revision: db5e189b3ba5996a91635e8ddf1396ed2018c698
      Branch: master
    Compiler: GNU 4.9.3
     Options: -std=gnu++98 -Wno-unused-local-typedefs -ggdb -Wall -Werror -fPIC
    Platform: linux2 3.16.0-4-amd64
        Bits: 64
        Mode: Debug
************************************ System ************************************
         CPU: Intel(R) Core(TM) i7-3630QM CPU @ 2.40GHz
      CPU ID: GenuineIntel Family 6 Model 58 Stepping 9
        CPUs: 1
      Memory: 1.45GiB
 Free Memory: 44.89MiB
     Threads: POSIX_THREADS
  OS Version: 4.2
 Has Battery: true
  On Battery: false
  UTC Offset: 3
         PID: 1986
         CWD: /home/virtualscience/Desktop/3dbg
********************************************************************************
QMetaObject::connectSlotsByName: No matching signal for on_addTool()
QMetaObject::connectSlotsByName: No matching signal for on_editTool(uint)
QMetaObject::connectSlotsByName: No matching signal for on_removeTool(uint)
Opening /usr/share/doc/camotics/examples/camotics/camotics.xml
Tool Path bounds ((0,0,-1), (119,16,3)) dimensions (119,16,4)
Computing surface bounded by ((-3,-3,-3), (122,18,0)) at 0 grid resolution
Time: 2,00 secs Triangles: 0 Triangles/sec: 0,00 Resolution: 0

jcoffland, 2016-05-19:
I believe the Debian package works on Ubuntu 15.10.

yaroslavVl, 2016-05-19:
I will try it

jcoffland, 2016-05-24:
How did it go?

resw, 2016-05-30:
Debian works. Thank you :D

renexdev, 2016-09-08:
Same problem with camotics_1.0.6_amd64.deb on ubuntu 15.10. I'll check with a previous camotics version.

closed 2017-03-26 by jcoffland