CauldronDevelopmentLLC/CAMotics #177: 4-axis simulation

kylotron, 2016-02-22:
Hi all...

i use BlenderCAM for generating gcode for 3 e 4th axis, and simulate my gcode for 3 axis with CAMotics.
But for 4th axis?

jcoffland, 2016-02-23:
I have been thinking about this and 5-axis simulation.  One of the issues of 4 and 5 axis simulation is that you must begin to consider the construction of the machine.  That is unless the GCode uses tool vectors, which most CAM programs do not generate.

How does the 4th axis on your machine work?  More specifically which axis does it rotate around?  Rotations around either the X or Z axis are typical.  A turntable like rotation is around the Z-axis and a lathe like rotation is around the X or Y axis although it could be on the diagonal in the XY plane but that seems unlikely.

kylotron, 2016-02-23:
My 4th axis is alligned on X axis (major axis for my cnc). I use LinuxCNC 2.6 for controlling my machine!

jcoffland, 2016-02-23:
Do you have full rotation or is it a trunnion with a limited range?

kylotron, 2016-02-24:
Full rotation with a step motor (and gear reduction 1:35)!