CauldronDevelopmentLLC/CAMotics #178: linux mint can't start cabotix, missing library: libstdc++

hyOzd, 2016-03-03:
I've installed CAMotics via [this](http://camotics.org/releases/public/release/camotics/debian-testing-64bit/v1.0/camotics_1.0.6_amd64.deb) debian package on Linux Mint 17.3 . When I run `camotics` I get this error in the command line:

    camotics: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by camotics)

I have this version of `libstdc++6` package installed:

    4.8.4-2ubuntu1~14.04.1

jcoffland, 2016-03-03:
You need at least ``libstdc++.so.6.0.20`` for ``GLIBCXX_3.4.20`` support.  This library is provided by ``libstdc++6`` v4.9.0.  Since you have v4.8.4 you are one minor version behind.  v4.8.4 provides C++ ABI support only up to ``GLIBCXX_3.4.19``.

jcoffland, 2016-03-03:
As a work around you could either try running the Ubuntu package or building from source.

hyOzd, 2016-03-03:
Hmm, thanks for info. I think it would be good idea to add this dependency into the debian package.

jcoffland, 2016-03-03:
Good point.  Even better would be to get CAMotics in to Debian.  I've been hoping to find someone to help me with this.

jcoffland, 2017-05-17:
Here's a solution for this problem in Ubuntu/Mint:

```
sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt-get update
sudo apt-get install gcc-4.9
sudo apt-get upgrade libstdc++6
```

You can check if you have ``GLIBCXX_3.4.20`` like this:

```
strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBCXX
```

chenz16, 2017-09-30:
The following [solution](https://askubuntu.com/questions/575505/glibcxx-3-4-20-not-found-how-to-fix-this-error) works for me: 

Follow these steps to solve the issue:

go to the right location and backup your current anaconda2 shortcut (change its name so it isn't overwritten):
cd ~/anaconda2/lib
mv -vf libstdc++.so.6 libstdc++.so.6.old
create a new shortcut using the ln command (I am assuming that I am in the previous location ~/anaconda2/lib):
ln -s /usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./libstdc++.so.6
restart spyder / other interface you use

------

jcoffland, 2019-02-09:
v1.2.0 now requires GLIBCXX_3.4.21 so this issue will not be fixed.  The only viable solution is to get a CAMotics source package accepted into Debian.

closed 2019-02-09 by jcoffland