CauldronDevelopmentLLC/CAMotics #180: camotics 1.0.7 ugly arcs

yaroslavVl, 2016-03-08:
Why arcs are so terrible?
[image]

jcoffland, 2016-03-09:
This does not look right.  Did you make any code changes in order to get it to compile in Ubuntu 15.10?

hyOzd, 2016-03-09:
I'm seeing the same thing. Linux Mint here. No code changes.

I also had some strange results with the 3d models, maybe its related. I will try to post a screenshot tomorrow.

jcoffland, 2016-03-09:
@hyOzd Are you using an official release or one you built from source?

hyOzd, 2016-03-09:
I've built from git source.

yaroslavVl, 2016-03-10:
No changes.  I followed the instruction step by step ( "sudo apt-get update", "sudo apt-get upgrade" before it).

jcoffland, 2016-03-10:
If you run ``camotics`` from the command line do you see any errors or warnings?  I'm running pretty much the same code and I don't see this problem.

hyOzd, 2016-03-10:
@jcoffland in my case I don't see any warnings or errors.

yaroslavVl, 2016-03-10:
there are no errors.
last ubuntu 15.10 running from oracle virtual box machine,
installing instructions from camotics web site.
*********************************** CAMotics ***********************************
     Website: http://camotics.org/
   Copyright: (c) 2011-2015
      Author: Joseph Coffland <joseph@camotics.org>
Organization: Cauldron Development LLC
        Args: 
      Config: <none>
************************************ Build *************************************
     Version: 1.0.7
        Date: Mar 8 2016
        Time: 15:02:17
  Repository: Git
    Revision: 084638a797d3200c1dd19c51ac33fabe45d157cb
      Branch: master
    Compiler: GNU 5.2.1 20151010
     Options: -std=gnu++98 -O3 -funroll-loops -ffast-math -mfpmath=sse
              -fno-unsafe-math-optimizations -msse2 -fPIC
    Platform: linux2 4.2.0-16-generic
        Bits: 64
        Mode: Release
************************************ System ************************************
         CPU: Intel(R) Core(TM) i7-3630QM CPU @ 2.40GHz
      CPU ID: GenuineIntel Family 6 Model 58 Stepping 9
        CPUs: 1
      Memory: 1.45GiB
 Free Memory: 729.46MiB
     Threads: POSIX_THREADS
  OS Version: 4.2
 Has Battery: true
  On Battery: false
  UTC Offset: 3
         PID: 1514
         CWD: /home/virtualscience
********************************************************************************
Opening /usr/share/doc/camotics/examples/camotics/camotics.xml
Tool Path bounds ((0,0,-1), (119,16,3)) dimensions (119,16,4)
Computing surface bounded by ((59.5,7.5,-1.5), (59.5,7.5,-1.5)) at 0 grid resolution
Time: 0,00 secs Triangles: 0 Triangles/sec: 0,00

misan, 2016-05-24:
Same problem here with 1.0.4 & 1.0.6 (both installed from your .deb) on Ubuntu 14.04 
[image]

misan, 2016-05-24:
Adding terminal output just in case is useful for diagnosing the problem:

*********************************** CAMotics ***********************************
     Website: http://camotics.org/
   Copyright: (c) 2011-2015
      Author: Joseph Coffland <joseph@camotics.org>
Organization: Cauldron Development LLC
        Args: 
      Config: <none>
************************************ Build *************************************
     Version: 1.0.6
        Date: Oct 28 2015
        Time: 10:38:49
  Repository: Git
    Revision: db5e189b3ba5996a91635e8ddf1396ed2018c698
      Branch: master
    Compiler: GNU 4.2.1 Compatible Ubuntu Clang 3.0 (tags/RELEASE_30/final)
     Options: -std=gnu++98 -O3 -funroll-loops -msse2 -fPIC
    Platform: linux2 3.16.0-4-amd64
        Bits: 64
        Mode: Release
************************************ System ************************************
         CPU: Intel(R) Core(TM) i5-4570 CPU @ 3.20GHz
      CPU ID: GenuineIntel Family 6 Model 60 Stepping 3
        CPUs: 4
      Memory: 7.66GiB
 Free Memory: 217.43MiB
     Threads: POSIX_THREADS
  OS Version: 3.13
 Has Battery: false
  On Battery: false
  UTC Offset: 2
         PID: 20709
         CWD: /home/misan
********************************************************************************
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x0
QMetaObject::connectSlotsByName: No matching signal for on_addTool()
QMetaObject::connectSlotsByName: No matching signal for on_editTool(uint)
QMetaObject::connectSlotsByName: No matching signal for on_removeTool(uint)
Opening /usr/share/doc/camotics/examples/camotics/camotics.xml
Tool Path bounds ((0,0,-1), (119,16,3)) dimensions (119,16,4)
Computing surface bounded by ((-3,-3,-3), (122,18,0)) at 0 grid resolution
Time: 2,00 secs Triangles: 0 Triangles/sec: 0,00 Resolution: 0

jcoffland, 2016-05-24:
I think I solved it.  Does the problem go away if you run like this?:

    LC_NUMERIC=C camotics

In fact, if I force a locale that uses commas instead of periods in numbers I get the same messed up arcs you are seeing.  This was actually already fixed in the [source code](https://github.com/CauldronDevelopmentLLC/CAMotics/commit/9533c212fa59d0895eb07b2a416beb1fce8ae3a3).  Future versions of CAMotics wont require manually setting the locale.

If this doesn't work then run ``locale -a`` and make sure it lists ``C`` as one of your system locales.

misan, 2016-05-24:
Exactly, using it as you suggest arcs look perfect. It is solved for me.

jcoffland, 2016-05-24:
Awesome!  Glad we finally figured it out.  Thanks for being persistent.  This locale problem has caused a lot of problems I should have suspected it sooner.

jcoffland, 2016-05-24:
Actually I'm going to leave this open until I make a release which includes the fix.

karelv, 2016-09-14:
out of the box it looks ugly, with the 'LC' trick, it works nice!

closed 2017-03-26 by jcoffland