CauldronDevelopmentLLC/CAMotics #183: Error running camotics on ubuntu 16.04

cjdg, 2016-05-06:
When i start camotics i get the following error:

ERROR:Exception: SmartPointer: Can't dereference a NULL pointer!

jcoffland, 2016-05-06:
Which package did you install?

cjdg, 2016-05-15:
ubuntu precise 64, there isnt for the newest ubuntu

jcoffland, 2016-05-16:
Does this one work any better?: http://camotics.org/releases/public/release/camotics/debian-testing-64bit/v1.0/camotics_1.0.6_amd64.deb

cjdg, 2016-05-16:
Yes it works!!! im testing it right now..

jcoffland, 2016-05-17:
That's great!

jferguson4, 2016-08-27:
I also found that the deb package works fine on 16.04 where the Ubuntu didn't.  Only glitch seems to be that tool diameter on simulation doesn't scale with tool using tool and workpiece in inches. more detail if it would be helpful. This is a very useful program especially for those of us who code on our couches and have no other way to look at what we've done before going downstairs to the less comfortable place to run it on LinuxCNC.  contribution coming.

thanks john

jcoffland, 2016-08-27:
If someone could try installing and running the debug version of the Ubuntu package that would be helpful in diagnosing the original problem.

http://camotics.org/releases/public/debug/camotics/ubuntu-precise-64bit/v1.0/camotics_1.0.6_amd64.deb

If you do so, please post the stack dump after ``ERROR:Exception: SmartPointer: Can't dereference a NULL pointer!`` here.

mage7223, 2016-09-15:
I had the same problem and I installed the debug version from @jcoffland and this is the output...

`*********************************** CAMotics ***********************************
     Website: http://camotics.org/
   Copyright: (c) 2011-2015
      Author: Joseph Coffland <joseph@camotics.org>
Organization: Cauldron Development LLC
        Args: 
      Config: <none>
************************************ Build *************************************
     Version: 1.0.6
        Date: Oct 28 2015
        Time: 10:38:49
  Repository: Git
    Revision: db5e189b3ba5996a91635e8ddf1396ed2018c698
      Branch: master
    Compiler: GNU 4.2.1 Compatible Ubuntu Clang 3.0 (tags/RELEASE_30/final)
     Options: -std=gnu++98 -Werror -fPIC
    Platform: linux2 3.16.0-4-amd64
        Bits: 64
        Mode: Debug
************************************ System ************************************
         CPU: AMD A6-4400M APU with Radeon(tm) HD Graphics
      CPU ID: AuthenticAMD Family 21 Model 16 Stepping 1
        CPUs: 2
      Memory: 5.30GiB
 Free Memory: 1.96GiB
     Threads: POSIX_THREADS
  OS Version: 4.4
 Has Battery: true
  On Battery: false
  UTC Offset: -4
         PID: 2602
         CWD: /home/krogers/Downloads
********************************************************************************
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x0
QMetaObject::connectSlotsByName: No matching signal for on_addTool()
QMetaObject::connectSlotsByName: No matching signal for on_editTool(uint)
QMetaObject::connectSlotsByName: No matching signal for on_removeTool(uint)
terminate called after throwing an instance of 'std::bad_cast'
  what():  std::bad_cast
Aborted (core dumped)
`

jcoffland, 2016-09-15:
Can you send me the core file it dumped?  It should be a file named ``core`` or something like ``core.####`` where ``####`` is the process id.  If you can upload to say Google Drive and share a link that would be very helpful.

jcoffland, 2016-09-25:
This is a duplicate of #174.

closed 2016-09-25 by jcoffland