CauldronDevelopmentLLC/CAMotics #184: change path color for different spindle speeds?

PeterGrace, 2016-05-19:
Similar to request #176, would it be possible to add the ability to alter the tool path color for areas where the spindle speed has changed?  I use CAMotics to verify my gcode patterns before sending them to my laser engraver, which uses spindle speed to alter the intensity of the laser.  When I preview images generated from a raster source, the toolpath moves along the entire area, line by line, producing a solid-green display in camotics.  You can only tell the spindle speed has changed by watching the right-side spindle-speed readout, which makes it difficult to visualize the final product via the camotics screen.

jcoffland, 2016-05-19:
That's a good idea and not too difficult to implement.

jcoffland, 2019-02-09:
This is implemented in the new v1.2.0 release.

closed 2019-02-09 by jcoffland