CauldronDevelopmentLLC/CAMotics #188: tool(1) ignored

DougCoffland, 2016-06-08:
If tool 1 is the first tool used, the resulting exported g-code does not prompt for the first tool.  i.e. no M6 T1 statement is generated.  This is causing me problems because LinuxCNC assumes the last tool used is the correct tool and begins to cut with the last tool.  The result is that the first time you run the g-code, it will correctly run with tool 1 and the switch to tool 2.  After that, it assumes that tool 2 is the correct tool and proceeds to run the whole program with tool 2.

The problem does not occur if you use tool 1 first and then use tool 1 later in the program.

jcoffland, 2016-06-08:
I assume this is a TPL issue.  Do you have example code?

imtaurus94, 2016-06-26:
Not all controlers need the M6 after the tool call (ex.: T1). There's gonna be a need for a customizing function in order to export G code based on the controler that's gonna read it.

jcoffland, 2016-06-26:
This would be solved by the *machine descriptions* feature which would, among other things, allow for specifying machine specific tool change routines.

closed 2017-03-26 by jcoffland