CauldronDevelopmentLLC/CAMotics #190: File names with non-english characters fail

jcoffland, 2016-08-27:


CitizenOfDreams, 2016-08-28:
Clarification: when you try to add a G-code file with international characters in the name (Russian in my case), the file name in displayed incorrectly and the file contents shows empty. See attached screenshot.

[image]

jcoffland, 2016-08-28:
Thanks, I'll look into it.

closed 2017-03-26 by jcoffland

CitizenOfDreams, 2017-03-27:
Sorry Jeff, but you closed the issue a bit prematurely. Files with Unicode names still don't work, at least for me.
- The file name is displayed correctly now;
- The file contents is displayed correctly;
- But the Simulation View tab stays empty.

jcoffland, 2017-03-27:
Ok, so this was only partially fixed.  I going to open a new issue for the remaining item.