CauldronDevelopmentLLC/CAMotics #196: CAMotics fails to build on Debian Testing

ksuprynowicz, 2016-11-08:
Hi!
I've tried to build the most recent version of CAMotics on Debian, but it fails due to not being able to find 'v8-version.h' file (the most recent library version is in Debian Testing is 3.14.5.8-8.1). With earlier versions everything worked perfectly. What libv8 version is required?

ksuprynowicz, 2016-11-08:
When I checkout cbang version from 27 September (last commit is febdb6977e4d38c943b2c6bb18298f9235527133) everything compiles and works perfectly.

jcoffland, 2016-11-08:
Sorry, I've been working on some code to use a newer version of libv8.  I had not intended to commit theses changes to the ``master`` branch.  I've fixed this issue.  Please try building from the ``master`` branch again.

closed 2016-11-08 by jcoffland

ksuprynowicz, 2016-11-08:
Thanks! Works perfectly now :)

jcoffland, 2016-11-08:
And thanks for the donation.  Much appreciated.