CauldronDevelopmentLLC/CAMotics #198: Support opening STL & DXF files

jcoffland, 2016-11-30:
This feature would open up a dialog asking the user for a handful of configuration options to turn a 2D STL or DXF in to a GCode program directly in CAMotics.

martonmiklos, 2016-12-06:
Hi there!

Are you planning to integrate toolpath generation (CAM) to the software?
For dealing with DXF files I could recommend this library:
http://www.ribbonsoft.com/en/what-is-dxflib

It is written by teh QCAD guys.

jcoffland, 2016-12-07:
Yes.  I did know about that library.  CAMotics already has a DXF lib.  You can see it here: https://github.com/CauldronDevelopmentLLC/CAMotics/tree/master/src/camotics/dxf  Thanks for the suggestion.

martonmiklos, 2016-12-07:
Ah sorry, I have not checked the code. 
Do you have any plans to integrate toolpath generation to the software?

jcoffland, 2016-12-09:
Yes, that's what this issue is about.

jcoffland, 2017-01-18:
Initially what I have in mind for DXF import is very simple.  You open a .dxf file and are then presented with a wizard type dialog.  You are presented with a list of DXF layers which you may:

  * Inset cut
  * Outset cut
  * Cut on
  * Pocket
  * Island (i.e. polygons inside a pocket which should not be cut.

You can also set cutting the depth, step down, feed, speed, tool, etc.  When you hit OK it will generate the GCode and simulate.  This is very similar to the style of pyCAM.

This may not be the most advanced interface but it should cover a large number of common use cases and make CAMotics much more useful for many people.

tpimh, 2022-02-25:
Any ideas on STL import?

jcoffland, 2022-02-25:
It's not yet supported.