CauldronDevelopmentLLC/CAMotics #200: Add qmake based .pro project file

martonmiklos, 2016-12-10:
I used to use QtCreator for developing Qt apps, and I think it would be useful to have a pro file for being able to build with the Qt's own build system. 

I have tried to get the setup working, here is where I got so far:
I have tried to link with the cbang in static version with -l:cbang.a, but I got :
undefined reference to symbol 'dlclose@@GLIBC_2.2.5'

The same issue persist even if I added -ldl and -lm flags. 
So I have tried to build the static version, which will link if I rename the so file to libcbang0.so, but the linker will miss the re2 library and event library with errors like this:

error: undefined reference to `evhttp_set_bevcb'
error: undefined reference to `re2::RE2::RE2(std::string const&)'

If you have any guidance please let me know! 
Thanks in advance!

martonmiklos, 2016-12-11:
Okay I got it working by linking dynamically to the shared version of all dependent libraries.
Are you willing to accept a PR for adding a pro file?

jcoffland, 2016-12-12:
Sure.  Open a PR and I'll accept it.

lnxbil, 2017-10-15:
Any news on that qmake file? I haven't seen a PR nor the file in GIT.

martonmiklos, 2017-10-15:
Aye, I have forgotten about it. Give me a few days to for it!

lnxbil, 2017-10-15:
Thank you! Maybe you could try to get some of your commits also into upstream. I really like your reload-Commit. That was one thing I was going to implement myself, because I also need it. Thank's to you, I do not need to write it myself.

yubobo0622, 2020-12-23:
Hi martonmiklos :
Reading this thread of CAMotics. I want to use QtCreator for developing Qt apps based CAMotics.
But i cannot find QtCreatot pro file in CauldronDevelopmentLLC/CAMotics.
Could you show me the pro file of CAMotics? or sending to my email: yubobo0622@126.com
Thanks to you vary much.