CauldronDevelopmentLLC/CAMotics #203: MDI windows?

martonmiklos, 2016-12-28:
Joseph

In the #198  issue you have mentioned that toolpath generation is planned to be integrated. 
What is the game plan for implementing it?

I would recommend to use similar MDI windows based layout like the VCarve does. 

http://www.shopbottools.com/images/Products/VCarveSample.png

jcoffland, 2017-01-18:
MDI stands for "Manual Data Input" which usually means manually entering GCodes.  The screenshot you linked above is for drawing and creating toolpaths which is something different.  Regardless I don't envision implementing drawing primitives in CAMotics.  There are already many programs which do this well.  Inkscape or LibreCAD, for example.  What I do envision is CAMotics importing DXF or SVG files and converting them in to tool paths (i.e. GCode).  This might involve some of the functions depicted on the right side of the above screenshot.  It probably would also allow for transforms such as rotate, scale and translate as well as array generation, i.e. repeated cuts of the same part.  This is covered by #198.

closed 2017-01-18 by jcoffland

martonmiklos, 2017-01-18:
Okay, we have misunderstood each other. 
I referred to the MDI window handling mechanism which stands for Multiple Document Interface:
https://msdn.microsoft.com/en-us/library/windows/desktop/ms632591(v=vs.85).aspx
When basically you have "windows in windows".
Qt supports it natively:
http://doc.qt.io/qt-5/qtwidgets-mainwindows-mdi-example.html

If we are going to add multiple features to the CAMotics it might be a good way to organize the UI.

jcoffland, 2017-01-18:
Oh, I forgot about that MDI.  I think this could be handled with DXF layers.  It would be easy to allow selecting which layers to cut.  E.g. some layers are offset inside, some outside, some are pockets, some islands, some ignored, etc.  This could all be done with out creating another drawing program.  See my comments on #198.