CauldronDevelopmentLLC/CAMotics #206: ases positive direction should be made clear

greyltc, 2017-01-27:
The rendering of the axes indicator in the simulation view is not clear on which direction is positive.
[image]

jcoffland, 2017-01-30:
How should this be done?  Could maybe float a plus sign near the positive side.

greyltc, 2017-01-30:
I'd just delete the arrows that point along negative directions. That should remove any ambiguity.
Here are some examples from freecad's axis/origin indicators:
[image]
[image]

commit 03703a6a (2023-08-10): Only show positive axis directions.  Closes #206
closed 2023-08-10 by jcoffland