CauldronDevelopmentLLC/CAMotics #207: tplang: Executable Stack Markings

apinsard, 2017-02-17:
Hi,

I'm currently packaging CAMotics for Funtoo (derivative from Gentoo) and the QA system reports an issue with "executable stack markings" in tplang.

```
 * QA Notice: The following files contain writable and executable sections
 *  Files with such sections will not work properly (or at all!) on some
 *  architectures/operating systems.  A bug should be filed at
 *  https://bugs.gentoo.org/ to make sure the issue is fixed.
 *  For more information, see:
 * 
 *    https://wiki.gentoo.org/wiki/Hardened/GNU_stack_quickstart
 * 
 *  Please include the following list of files in your report:
 *  Note: Bugs should be filed for the respective maintainers
 *  of the package in question and not hardened@g.o.
 * RWX --- --- usr/bin/tplang
```

I must say I have no idea what it is, but according to Gentoo wiki here: https://wiki.gentoo.org/wiki/Hardened/GNU_stack_quickstart I think it's an upstream issue.

> GCC generates code to be executed on the stack when it implements a [trampoline for nested functions](http://gcc.gnu.org/onlinedocs/gccint/Trampolines.html). To remove the need for an executable stack in this case, it is necessary to rewrite the code another way. Sometimes this is relatively easy, other times not.

Here is the output of `scanelf -qeR .` after compilation phase:

    RWX --- ---  ./tplang
    RWX --- ---  ./.sconf_temp/conftest_43

And here is the full build log, if this may help:

```
>>> Compiling source in /var/tmp/portage/media-gfx/CAMotics-1.1.0/work/CAMotics-1.1.0-release ...
scons -j4 -C /var/tmp/portage/media-gfx/CAMotics-1.1.0/work/CAMotics-1.1.0-release
scons: Reading SConscript files ...

scons: warning: QT5DIR variable not defined, and detected moc is for Qt 4.8.7
File "/var/tmp/portage/media-gfx/CAMotics-1.1.0/work/CAMotics-1.1.0-release/config/qt5/__init__.py", line 401, in _detect

scons: warning: QT4DIR variable is not defined, using moc executable as a hint (QT4DIR=/usr)
File "/var/tmp/portage/media-gfx/CAMotics-1.1.0/work/CAMotics-1.1.0-release/config/qt4/__init__.py", line 394, in _detect
  Compiler: gcc (gnu)
  Platform: posix
  Mode: gnu
  Arch: x86
Checking for C library pthread... yes
running with -j 4
Checking for C header file zlib.h... yes
Checking for C library z... yes
Checking for C header file bzlib.h... yes
Checking for C library bz2... yes
Checking for C header file expat.h... yes
Checking for C library expat... yes
Checking for C header file pthread.h... yes
Checking for C library pthread... yes
Checking for C library dl... yes
Checking for C library sqlite3... yes
Checking for C header file sqlite3.h... yes
Checking for C function sqlite3_backup_init()... yes
Checking for C header file event.h... yes
Checking for C library event... yes
Checking for C++ header file re2/re2.h... yes
Checking for C library re2... yes
Checking for C library leveldb... no
Checking for C header file mysql/mysql.h... yes
Checking for C library mysqlclient... yes
Checking for C function mysql_real_connect_start()... no
Checking for boost version >= 1.40...yes
Checking for C++ header file boost/version.hpp... yes
Checking for C++ header file boost/iostreams/stream.hpp... yes
Checking for C++ header file boost/ref.hpp... yes
[... 308 lines cut ...]

 * Final size of build directory: 73398 KiB
 * Final size of installed tree: 16 KiB

 * QA Notice: The following files contain writable and executable sections
 *  Files with such sections will not work properly (or at all!) on some
 *  architectures/operating systems.  A bug should be filed at
 *  https://bugs.gentoo.org/ to make sure the issue is fixed.
 *  For more information, see:
 * 
 *    https://wiki.gentoo.org/wiki/Hardened/GNU_stack_quickstart
 * 
 *  Please include the following list of files in your report:
 *  Note: Bugs should be filed for the respective maintainers
 *  of the package in question and not hardened@g.o.
 * RWX --- --- usr/bin/tplang

strip: x86_64-pc-linux-gnu-strip --strip-unneeded -R .comment -R .GCC.command.line -R .note.gnu.gold-version
   usr/bin/camotics
   usr/bin/camsim
   usr/bin/gcodetool
   usr/bin/tplang
   usr/lib64/libCAMotics.a
   usr/lib64/libclipper.a
```

jcoffland, 2017-02-17:
I tested all of the object files that make up tplang and none of them lack the ``.note.GNU-stack`` symbol so the problem must be in one of the libraries linked to tplang.  The Javascript library is the only library unique to tplang.  CAMotics can use either v8 (Google) or Chakra (Microsoft) Javascript libraries.  I suspect Chakra is the problem.  Try linking to libv8 instead.

jcoffland, 2019-01-19:
http://docs.flatpak.org/

closed 2019-01-19 by jcoffland