CauldronDevelopmentLLC/CAMotics #209: tplang: Library QtOpenGL.framework/Versions/4/QtOpenGL not found on macOS

tcurdt, 2017-03-02:
Fresh install on 10.12.3 (16D32) gives this on startup:

```
Process:               tplang [85900]
Path:                  /Applications/CAMotics.app/Contents/MacOS/tplang
Identifier:            tplang
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        camotics [85897]
Responsible:           tplang [85900]
User ID:               501

Date/Time:             2017-03-02 11:53:40.915 +0100
OS Version:            Mac OS X 10.12.3 (16D32)
Report Version:        12
Anonymous UUID:        C5809969-9982-626B-185C-C71743BF1FC0

Sleep/Wake UUID:       E629BEDC-83C7-4535-A354-D43533C61504

Time Awake Since Boot: 520000 seconds
Time Since Wake:       3700 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: QtOpenGL.framework/Versions/4/QtOpenGL
  Referenced from: /Applications/CAMotics.app/Contents/MacOS/tplang
  Reason: image not found

Binary Images:
       0x100000000 -        0x1007d0ed7 +tplang (???) <3A32233B-11DE-35A3-A278-0606A21528BC> /Applications/CAMotics.app/Contents/MacOS/tplang
       0x1008ed000 -        0x100906ffb  libexpat.1.dylib (15) <617B8B7A-A0AB-37D6-AFCA-F90F7F14E455> /usr/lib/libexpat.1.dylib
       0x103ade000 -        0x103b1b267  dyld (421.2) <947FC440-80F9-32F7-A773-6FC418FE1AB7> /usr/lib/dyld
    0x7fff7643f000 -     0x7fff768d9fff  com.apple.CoreFoundation (6.9 - 1348.28) <A40AA224-7A50-3989-95D0-5A228A0E2FAF> /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
    0x7fff77304000 -     0x7fff77304fff  com.apple.CoreServices (775.9.7 - 775.9.7) <A5C444F3-408B-3062-AF4B-BF8CD919F221> /System/Library/Frameworks/CoreServices.framework/Versions/A/CoreServices
    0x7fff78421000 -     0x7fff784b6ff7  com.apple.framework.IOKit (2.0.2 - 1324.30.13) <163BE7FA-B29A-348F-8B5F-E301F2E8C964> /System/Library/Frameworks/IOKit.framework/Versions/A/IOKit
    0x7fff7b4b9000 -     0x7fff7b4c7fff  com.apple.opengl (13.0.10 - 13.0.10) <B11A2E1B-4B1D-3ECD-BADA-3508BE5775BF> /System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL
    0x7fff8a47b000 -     0x7fff8a47cff3  libSystem.B.dylib (1238) <9CB018AF-54E9-300F-82BE-81FE553C9154> /usr/lib/libSystem.B.dylib
    0x7fff8a5a6000 -     0x7fff8a5b4ff7  libbz2.1.0.dylib (38) <6FD3B63F-0F86-3A25-BD5B-E243F58792C9> /usr/lib/libbz2.1.0.dylib
    0x7fff8b5b8000 -     0x7fff8b700fe3  libsqlite3.dylib (253) <B5BA5C96-AB13-34A0-8237-DD52A0181DFE> /usr/lib/libsqlite3.dylib
    0x7fff8b75c000 -     0x7fff8b7acfff  libstdc++.6.dylib (104.1) <91F66BFD-F927-301F-B8F3-578A5CEA78F4> /usr/lib/libstdc++.6.dylib
    0x7fff8b920000 -     0x7fff8b931ff3  libz.1.dylib (67) <46E3FFA2-4328-327A-8D34-A03E20BFFB8E> /usr/lib/libz.1.dylib

Model: MacBookAir6,2, BootROM MBA61.0099.B26, 2 processors, Intel Core i7, 1.7 GHz, 8 GB, SMC 2.13f15
Graphics: Intel HD Graphics 5000, Intel HD Graphics 5000, Built-In
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x483943434E4E4E384A544D4C41522D4E544D
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x483943434E4E4E384A544D4C41522D4E544D
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x117), Broadcom BCM43xx 1.0 (7.21.171.68.1a5)
Bluetooth: Version 5.0.3f1, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SD0256F, 251 GB
USB Device: USB 3.0 Bus
USB Device: Card Reader
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Air, Apple Inc., 23.6
```

jcoffland, 2017-03-03:
So it's failing to load the QtOpenGL library.  Does CAMotics start at all?

jcoffland, 2017-03-27:
Any more info on this?

tcurdt, 2017-03-27:
Sorry, I didn't see the first comment.

Yes, it starts up, shows the main windows and then the crash report comes up.
Given the main window stays alive I assume it must me some other process/executable that fails.

Let me know if I can help to debug this further.

jcoffland, 2017-03-27:
A screenshot would be helpful  I'm interested to know if you seen any 3D graphics in the application or if it's blank.  It could be that the ``QtOpenGL`` library is included and working but it is incorrectly linked to ``tplang`` which shouldn't need it.

tcurdt, 2017-03-27:
These are the libraries that are being used:
```
$ otool -L tplang 
tplang:
	/usr/lib/libstdc++.6.dylib (compatibility version 7.0.0, current version 104.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1226.10.1)
	/usr/lib/libsqlite3.dylib (compatibility version 9.0.0, current version 216.4.0)
	/usr/lib/libexpat.1.dylib (compatibility version 7.0.0, current version 8.0.0)
	/usr/lib/libbz2.1.0.dylib (compatibility version 1.0.0, current version 1.0.5)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.5)
	/System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1258.1.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/System/Library/Frameworks/CoreServices.framework/Versions/A/CoreServices (compatibility version 1.0.0, current version 728.13.0)
	QtOpenGL.framework/Versions/4/QtOpenGL (compatibility version 4.8.0, current version 4.8.6)
	QtGui.framework/Versions/4/QtGui (compatibility version 4.8.0, current version 4.8.6)
	QtCore.framework/Versions/4/QtCore (compatibility version 4.8.0, current version 4.8.6)
```
and this is how it looks like:

[image]

So there are some 3D graphics.

jcoffland, 2017-03-27:
Ok, ``tplang`` is linked to the Qt libraries, which it does not need to be, and it's having a problem loading ``QtOpenGL.framework/Versions/4/QtOpenGL`` even though ``camotics`` must be loading this fine since it is able to display 3D graphics.  Looks like I need to release a new build that does not link the command line tools to the Qt libraries.

rahe6266, 2017-05-04:
Thanks for making this application availiable.

I am running macOS on old Macbook built 2009 (patched for running this Version macOS 10.12.4 on older Hardware).

  Systemversion:	macOS 10.12.4 (16E195)
  Kernel-Version:	Darwin 16.5.0

When starting the application via opening sample file box.tpl, the following message shows up in the lower window half.

dyld: Library not loaded: QtOpenGL.framework/Versions/4/QtOpenGL
  Referenced from: /Applications/CAMotics.app/Contents/MacOS/tplang
  Reason: Incompatible library version: tplang requires version 4.8.0 or later, but QtOpenGL provides version 4.5.0
Computing surface bounded by ((-32.8482,-2.77031,-13.4909), (32.8482,110.77,0.79092)) at 0.714092 grid resolution
Time: 3.91 secs Triangles: 72616 Triangles/sec: 18579.76

while when opening a .nc file the (like the aztec calendar sample ) the calculation seems to be running without message regarding QtOpenGL.

Tool Path bounds ((-202.115,-202.489,-12.8854), (201.882,202.311,20.32)) dimensions (403.997,404.8,33.2054)
Computing surface bounded by ((-203.75,-203.75,-15.715), (203.75,203.75,4.285)) at 0.5 grid resolution
Time: 2 mins 14 secs Triangles: 7118228 Triangles/sec: 53039.79

Thanks for considering.

zrichmond, 2017-08-18:
I'm hitting the same error running tplang on OS X, camotics runs fine for me.  Looks like the release 1.1.1 on 2017-02-08 is still the latest, so I thought I'd see if I can build from source. Then see if I can eliminate the QT dependency.   Currently I'm getting stuck building cbang. Here's the first of a couple related errors:

`src/cbang/event/PersistentEventManager.cpp:40:14: error: 'Event' is not a class, namespace, or enumeration
SmartPointer<Event::Event>` 

Which is strange since Event.h is included.  My C++ is pretty rusty so any help would be great.

jcoffland, 2017-08-18:
@zrichmond, this really should have been reported as a separate issue since it's only vaguely related.  However, did the change I just checked into cbang fix this problem?

zrichmond, 2017-08-18:
@jcoffland Fixed, thanks.  Yeah sorry about the cross commenting, I know better too.

jcoffland, 2017-08-18:
@zrichmond glad to hear it.

tcurdt, 2017-09-09:
Any news on this?

zrichmond, 2017-09-09:
I ended up battling compile errors with CAMotics on OS X which I wasn't able to nail down into an issue. So, I didn't get into working on a fix yet...

jcoffland, 2017-09-11:
I'm working on a new release which should fix this problem.  We are in the middle of Kickstarter campain so things are a bit crazy right now.  http://buildbotics.com/kickstarter

jcoffland, 2019-02-05:
Fixed in upcoming v1.2.0 release.

closed 2019-02-05 by jcoffland