CauldronDevelopmentLLC/CAMotics #210: G0 Default Machine Feed

MysticFrog, 2017-03-05:
1. 
Could you please add a default rapid feed rate option to the settings. and make all G01 moves prior to the F command instant?

The program should assume it is at first X,Y at the start of the G-Code not at some random other position causing a long feed at the start of the program

How can you calculate program time with out knowing the machine rapid rates?

2.
Code like this causes the gfx to show the part slowly moving up each time is is called

N010 G28.1 Z3 F500
N020 G92 Z0
N030 G00 Z0
N040 G92 Z0

Certain g-codes should be able to be ignored.

***** Great program.

jcoffland, 2017-03-06:
The machine always starts out at (0,0,0) but if you run consecutive GCodes the next one starts where the last one left off.

I plan to add a feature which lets you describe the machine characteristics in detail.  See #55.  Currently the rapid feed is fixed at 1 meter/min.

For 2. can you provide a more complete example.  If I run that code in CAMotics I don's see anything.

MysticFrog, 2017-03-10:
Hi Joseph,

Thanks for your reply.

1.

If I load this program (test_1.tap, attached) it literally takes 3 hours to perform the first move because it is looking at the F1 in this cam generated program G00 X35 Y149

Why it is there I don't know I cannot change the post, but G0 shouldn't be affected by a programmed F code.

If I add a feed rate to this line N0140 or remove the F1 it starts working normally

2.

I have attached another file test1_1.tap in this file there are multiple parts nested (this is for a profile cutting machine). At the beginning of each part the machine tests the height of the material then moves a set distance and resets its Z zero position. Your program (and MACH3) do not understand what is happening and interpret it as the z offset position slowly moving upwards.

I suspect this is not a bug and your graphical representation is showing as it should do which is why I would like a way to ignore certain G-codes

[cid:70385c7f-ef07-42d1-a3f2-0d9210e4d4f2]

Thanks again.

Andy

jcoffland, 2017-03-10:
Your attachment didn't come through.

MysticFrog, 2017-03-13:
Hi Joseph,

Try this.

Thanks

Andy

jcoffland, 2017-03-14:
That still did not work.  Attachments sent via Outlook do not work on Github issues.  You have to go to the issue page: https://github.com/CauldronDevelopmentLLC/CAMotics/issues/210 and attach or paste your files there.

MysticFrog, 2017-03-15:
[test_1.txt](https://github.com/CauldronDevelopmentLLC/CAMotics/files/843080/test_1.txt)
[test1_1.txt](https://github.com/CauldronDevelopmentLLC/CAMotics/files/843082/test1_1.txt)

Now I understand why it wasn't working

I had to change the ext to .txt to get it to upload.

jcoffland, 2017-03-15:
I don't see any problem with ``test_1.txt`` other than it has rapid moves in the material.  What version of CAMotics are you using?

[image]

This is what I see with ``test1_1.txt``.

[image]

First off M22 is not supported.  WikiPedia says M22 means "Mirror, Y-axis".

The stair stepping has to do with how [G28.1](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g28-g28.1) and [G92](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g92) are interpreted.  G28.1 simply saves a predefined position.  You would have to call G28 use that saved position.  The calls to G92 set a coordinate system offset for the z-axis.  The stair stepping is caused by this code:

    G00 Z2.5
    . . .
    G92 Z0

It keeps telling the simulator to reinterpret Z2.5 as Z0.  This adds 2.5mm to the tool height each time.

There are many different types of GCode.  Likely your machine understands a different dialect than CAMotics or MACH3 support.

MysticFrog, 2017-03-16:
try this program 
[square1_1.txt](https://github.com/CauldronDevelopmentLLC/CAMotics/files/846825/square1_1.txt)
[image]

I understand your comments . this is why I asked for a way to ignore certain G-Codes 
This machine obviously uses these G-codes in a non conventional way.

and yes M22 is customised for a MACH3 plugin.

jcoffland, 2017-03-27:
I get the same simulation you show above.  What's wrong with it?

jcoffland, 2017-03-27:
Never mind,  I see now that when you run the tool path playback the tool and path disappear.  I'm going to open a new issue for this.  Thanks for the clear test case.

MysticFrog, 2017-03-28:
Hi Joseph.

I don't quite understand what you can see

The screen shot I posted shows a program which has a runtime of over one day to cut a 3 minute part

I think it's because the first G0 is not at G0 speed.

jcoffland, 2017-03-28:
Ahh, I did not see the 1 day 9 minute time.  Also, notice that the tool path and tool have disappeared from the view altogether.  There's something more going on here.  Probably a divide by zero in the visualization code.  That tends to make things disappear in OpenGL.

kfmut, 2020-01-10:
@jcoffland 

Hi,

I'm curious what is current status for specifying G0 feed rate. Is it possible to write rapid feed rates for individual axes to machine configuration file or something? Also how rapids are accounted in machining time estimation?

jcoffland, 2020-01-10:
Not yet.

kfmut, 2020-01-10:
@jcoffland 

So rapids are skipped in machining time estimation or some default value is used?

jcoffland, 2020-01-10:
A default value of 10,000/mm/min is used.  See #55.

kfmut, 2020-01-12:
Thank you for clarification!