CauldronDevelopmentLLC/CAMotics #211: Missing shared MIME-info file

srakitnican, 2017-03-08:
What are all the file types CAMotics can open?

So this is a standard shared MIME-info file that all desktops are following (as far as I know) when deciding on what application needs to open a specific file. The more files, the better, there is even a possibility of specifying a "weight" if more application can open the same file or specifying file types by "magic" bytes.

Example of what it may look like:
```
<?xml version="1.0" encoding="utf-8"?>
<mime-info xmlns="http://www.freedesktop.org/standards/shared-mime-info">
  <mime-type type="application/x-camotics-project">
    <sub-class-of type="text/plain"/>
    <comment>CAMotics Project</comment>
    <glob pattern="*.gcode"/>
    <glob pattern="*.tap"/>
    <glob pattern="*.tpl"/>
    <glob weight="30" pattern="*.xml"/>
...
  </mime-type>
</mime-info>
```

File goes into /usr/share/mime/packages/<name>.xml and it is required by the Fedora Packaging Guidelines.

srakitnican, 2017-03-09:
So I've put something together based on manual and look at the source, feel free to include it.

camotics.xml
```
<?xml version="1.0" encoding="utf-8"?>
<mime-info xmlns="http://www.freedesktop.org/standards/shared-mime-info">
  <mime-type type="application/x-camotics-project">
    <sub-class-of type="text/plain"/>
    <comment>CAMotics project file</comment>
    <glob pattern="*.gcode"/>
    <glob pattern="*.nc"/>
    <glob pattern="*.ngc"/>
    <glob pattern="*.tap"/>
    <glob pattern="*.tpl"/>
    <glob weight="30" pattern="*.xml" />
  </mime-type>
</mime-info>
```

I've encountered one warning in CAMotics.desktop file, please consider fixing it.
```
/builddir/build/BUILDROOT/camotics-1.1.1-1.fc25.x86_64/usr/share/applications/CAMotics.desktop: error: (will be fatal in the future): value "camotics.png" for key "Icon" in group "Desktop Entry" is an icon name with an extension, but there should be no extension as described in the Icon Theme Specification if the value is not an absolute path
```

jcoffland, 2017-03-09:
That looks good but will it associate all XML files with CAMotics?

I added issue #212.

srakitnican, 2017-03-09:
Yes, but based on default weight (50), any other application that does not have the weight attribute set or lower then 30 can take it. There is also an option to use "magic" attribute that looks into file for specific string or bit, so maybe that would be a better option. Example from libgeda.xml which is looking at the beginning of the file for string "v 201" or "v 200"

```
<?xml version="1.0" encoding="utf-8"?>
<mime-info xmlns="http://www.freedesktop.org/standards/shared-mime-info">
  <mime-type type="application/x-geda-schematic">
    <sub-class-of type="text/plain"/>
    <comment>gEDA circuit schematic</comment>
    <glob pattern="*.sch"/>
    <magic priority="70">
      <match value="v 201" type="string" offset="0"/>
      <match value="v 200" type="string" offset="0"/>
    </magic>
  </mime-type>
...
```

jcoffland, 2017-03-09:
All CAMotics XML files should have ``<camotics>`` at the beginning.  This could conceivably be preceded by ``<?xml version="1.0" encoding="utf-8"?>`` but CAMotics does not currently write project files this way.

srakitnican, 2017-03-10:
This should cover both cases, I've also divided filetypes into two groups to give it more details.:
```
<?xml version="1.0" encoding="utf-8"?>
<mime-info xmlns="http://www.freedesktop.org/standards/shared-mime-info">
  <mime-type type="application/x-camotics-nc">
    <sub-class-of type="text/plain"/>
    <comment>CAMotics Numerical Control file</comment>
    <glob pattern="*.gcode"/>
    <glob pattern="*.nc"/>
    <glob pattern="*.ngc"/>
    <glob pattern="*.tap"/>
    <glob pattern="*.tpl"/>
  </mime-type>
  <mime-type type="application/x-camotics-project">
    <sub-class-of type="text/plain"/>
    <comment>CAMotics project</comment>
    <magic priority="70">
      <match type="string" offset="0" value="&lt;camotics&gt;"/>
      <match type="string" offset="39" value="&lt;camotics&gt;"/>
    </magic>
    <glob pattern="*.xml" />
  </mime-type>
</mime-info>
```

If you think format of the xml header may change there is also an option to give it range, something like this should cover all cases:
`      <match type="string" offset="0:60" value="&lt;camotics&gt;"/>`

srakitnican, 2017-03-10:
I am happy with following
```xml
<?xml version="1.0" encoding="utf-8"?>
<mime-info xmlns="http://www.freedesktop.org/standards/shared-mime-info">
  <mime-type type="application/x-camotics-nc">
    <sub-class-of type="text/plain"/>
    <comment>CAMotics Numerical Control</comment>
    <glob pattern="*.gcode"/>
    <glob pattern="*.nc"/>
    <glob pattern="*.ngc"/>
    <glob pattern="*.tap"/>
    <glob pattern="*.tpl"/>
  </mime-type>
  <mime-type type="application/x-camotics-project">
    <sub-class-of type="text/plain"/>
    <comment>CAMotics project</comment>
    <magic priority="80">
      <match type="string" offset="0:60" value="&lt;camotics"/>
    </magic>
    <glob weight="30" pattern="*.xml" />
  </mime-type>
</mime-info>

```
 `weight="30"`, lower then default value,  assures that xml file extension is not associated with CAMotics if there is no string found. Also made the search string `<camotics` in case attributes are implemented in the future.

jcoffland, 2017-03-10:
This is a nice addition.  Thanks!  It will be in the next release.

srakitnican, 2017-03-10:
That's great!

Don't forget the MimeType in CAMotics.desktop.
```
MimeType=application/x-camotics-project;application/x-camotics-nc;
```

jcoffland, 2017-03-10:
Adding it now.  Thanks.

jcoffland, 2019-02-05:
Fixed in upcoming v1.2.0 release.

closed 2019-02-05 by jcoffland