CauldronDevelopmentLLC/CAMotics #212: CAMotics.desktop causes warning regarding icon filename

jcoffland, 2017-03-09:
```
/builddir/build/BUILDROOT/camotics-1.1.1-1.fc25.x86_64/usr/share/applications/CAMotics.desktop: error: (will be fatal in the future): value "camotics.png" for key "Icon" in group "Desktop Entry" is an icon name with an extension, but there should be no extension as described in the Icon Theme Specification if the value is not an absolute path
```

srakitnican, 2017-07-07:
Hi, it would be nice if this would be fixed before new release. Do you need a PR?

srakitnican, 2017-07-07:
So basically this is all that is needed:

```diff
--- CAMotics.desktop	2017-07-07 11:39:52.555162366 +0200
+++ CAMotics.desktop.new	2017-07-07 11:40:25.037005197 +0200
@@ -6,6 +6,6 @@
 Terminal=false
 Categories=Science
 Comment=Open-Source Simulation & Computer Aided Machining
-Icon=camotics.png
+Icon=camotics
 StartupNotify=true
 MimeType=application/x-camotics-project;application/x-camotics-nc;

```

jcoffland, 2017-07-07:
Done.

commit d815aafa (2017-07-07): Fixed desktop icon reference.  Closes #212
closed 2017-07-07 by jcoffland