CauldronDevelopmentLLC/CAMotics #215: Missing libv8-3.14.5 on debian 8 64

scorninpc, 2017-03-24:
Output of fresh .deb instalation from camotics site

> $ camotics 
> camotics: error while loading shared libraries: libv8.so.3.14.5: cannot open shared object file: No such file or directory
> $ apt-get install libv8-3.14.5
> ...
> ...
> $ camotics 

Sorry for less information, i'm new here. If need more help ask me

jcoffland, 2017-03-24:
Are you saying it worked after you installed ``libv8``?

scorninpc, 2017-03-24:
Yes, work fine after installing

jcoffland, 2017-03-27:
I've added a dependency on this package.

scorninpc, 2017-03-27:
I think it's not compile with. Download package and see the source dependency

http://i.imgur.com/9WZFHpO.jpg (can I send images here?)

jcoffland, 2017-03-27:
Right, it will be in the next release.

RonnieJHW, 2018-02-07:
I also had to separately add the libv8-3.14.5 to get camotics to work.  Using Ubuntu 16.04, 64 bit.

ipsod, 2018-05-19:
Having the same issue in Linux Mint 18.3 - same fix works.

sk8nfool, 2018-12-19:
This problem (missing libv8 dependency) is still present in camotics_1.1.1_amd64.deb. Manually installing libv8 resolves the problem.

jcoffland, 2018-12-19:
@sk8nfool  What Linux distro are you using that has a ``libv8`` package.  I'm on a recent Debian testing and it only has ``libv8-3.14.5``.

sk8nfool, 2018-12-19:
I'm currently using Ubuntu 18.04 but that will change in the near future.
Also - I checked what I have installed and it is libv8-3.14.5
Version 3.14.5.8-11ubuntu1 - someone is messing with the typical notation I
think.

jcoffland, 2019-02-05:
Fixed in upcoming v1.2.0 release.

closed 2019-02-05 by jcoffland

wulph111, 2020-03-19:
Installed v1.2.0 on a fresh Debian 10 just now, this does not appear to be fixed. Further, the required libraries don't seem to be available.

[image]

on4ain, 2020-10-04:
installed V1.2.0 on fresh Mint 20 and same issue:
[image]

cefn, 2021-11-14:
Similar issue on Focal Fossa LTS and libv8 doesn't have any installation candidates...

```
$ camotics
camotics: error while loading shared libraries: libv8.so.3.14.5: cannot open shared object file: No such file or directory
```

Laurent6012, 2022-07-11:
Hi,
Similar issue on Linux Lite 5.0 based on Ubuntu 20.04, fresh installed.
So, a solution here that worked for me :
Do a simlink of libv8.so, if exist the lib is in /usr/lib/x86-64-linux-gnu/.
`sudo ln -s /usr/lib/x86_64-linux-gnu/libv8.so /home/lolo/Bureau/`
I'm french so for you it would be certainly 'Desktop' not 'Bureau'
After that rename it in libv8.so.3.14.5 and move it back in the /usr/lib/x86-64-linux-gnu/ folder, the code : `sudo ln  mv /home/loulou/Bureau/libv8.so.3.14.5 /usr/lib/x86_64-linux-gnu/`
I renamed it directly on the desktop, for those who could not do that you can do a copy instead. No need to rename it but you'll must delete it after :
`sudo cp /home/loulou/Bureau/libv8.so /usr/lib/x86_64-linux-gnu/libv8.so.3.14.5`
Camotics will not see a difference with the simlink and it appears the lib is the right one for other distro, at least for Linux Lite 5.0.
Have fun

Symbian9, 2022-09-18:
> Fixed in upcoming v1.2.0 release.

@jcoffland, Dependencies still missing in latest `v1.2.0` release DEB, as it built to actually outdated Debian 9.x Stretch.

I just tested under *Debian 11.x Bullseye*, and there was missed packages required to install manually:

- `libv8.so.3.14.5`; Need to install `libv8` (>= 3.14.5) manually:
  - https://packages.debian.org/stretch/libv8-3.14.5 (***available only in Debian 9.x Stretch repo***)
- `libQt5WebSockets.so.5`; Need to install `libqt5websockets5`:
  - https://packages.debian.org/bullseye/libqt5websockets5

dwelch67, 2022-11-01:
$ sudo apt-cache search libv8
libnode-dev - evented I/O for V8 javascript (development files)
$ sudo apt-get install libnode-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libnode-dev is already the newest version (10.19.0~dfsg-3ubuntu1).
libnode-dev set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
$ camotics 
camotics: error while loading shared libraries: libv8.so.3.14.5: cannot open shared object file: No such file or directory
$ 
$ uname -a
Linux desktop 5.15.0-52-generic #58~20.04.1-Ubuntu SMP Thu Oct 13 13:09:46 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Linuxmint
Description:	Linux Mint 20.3
Release:	20.3
Codename:	una

20.3	Una  Ubuntu Focal

Ubuntu 20.04.5 LTS (Focal Fossa)

Aiph5u, 2023-05-29:
Hey all,

Debian 9 "[stretch](https://www.debian.org/releases/stretch/)" is now **archived**, the link above to libv8.3.14 is dead and is now:

https://archive.debian.org/debian/pool/main/libv/libv8-3.14/

or, into your /etc/apt/sources.list, include:

`deb http://archive.debian.org/debian/ stretch            main contrib non-free`

and do 'apt-get update'

holozone, 2023-06-07:
Same Problem here with camontics 1.2 on debian bullseye, camotics installed with the deb-packaged and gdebi, everything installs well but it do not start:
> camotics 
camotics: error while loading shared libraries: libv8.so.3.14.5: cannot open shared object file: No such file or directory

Aiph5u, 2023-06-07:
@jcoffland Perhaps this issue was closed too soon ;)

oarcher, 2023-06-10:
Same issue with ubuntu lunar 23.04.

gcanzalo, 2023-06-11:
Same with Debian Bookworm/Camotics 1.2.

gx-turbo, 2023-07-28:
Is there an action from camotics? It seems a frequent problem. Also with my latest ubuntu.

lukaspetr, 2023-12-06:
I encountered the problem: 

`camotics: error while loading shared libraries: libv8.so.3.14.5: cannot open shared object file: No such file or directory
`
when trying to run camotics, my system is Ubuntu 22.04, I installed already `libnode-dev` via apt-get.

barnabywalters, 2024-01-15:
Still happening on current Kubuntu after installing the .deb:

    camotics: error while loading shared libraries: libv8.so.3.14.5: cannot open shared object file: No such file or directory

Operating System: Kubuntu 23.10
KDE Plasma Version: 5.27.8
KDE Frameworks Version: 5.110.0
Qt Version: 5.15.10
Kernel Version: 6.5.0-14-generic (64-bit)

hesperaux, 2025-06-22:
Still a problem on up-to-date Debian 12.11 with libnode-dev installed.
CAMotics relies on the full semver'd soname (libv8.so.3.14.5), but this is no longer in the node release. There is a libnode.so provided, and libv8.so is a symlink to that. There is no longer a livb8.so.<anything> on Debian 12, and it appears to be the same on Ubuntu and others, based on the comments.
The following is a hack that will allow camotics to run:
`sudo ln -s libv8.so libv8.so.3.14.5`
But, obviously, this is a stop gap measure and not advisable, since it could break your system over time.

Even with this hack, I am still seeing symbol loading errors (not sure from which shared library they are expected):
/usr/bin/tplang: symbol lookup error: /usr/bin/tplang: undefined symbol: _ZN2v811HandleScopeC1Ev

This may be related to an on-launch tool path generation error.