CauldronDevelopmentLLC/CAMotics #222: Export GCode never uses G02/G03 for TPL arc()?

sheffieldnick, 2017-04-01:
Using TPL's `arc(x, y, z, angle)` always exports GCode with many linear G1 commands instead of 1-4 G2/G3 commands? Can this be a config option so that GCode files are much smaller if that particular CNC machine understands G2/G3 ?

Thanks

jcoffland, 2017-04-01:
There is a good reason for this.  TPL can do matrix transformations on the GCode.  This produces arcs which are not possible with G2/G3.  However, TPL could be made smarter so that it outputs G2/G3 when the matrix does not skew the arcs.  It could detect matrices which only scale, flip or translate arcs and modify the G2/G3 parameters as necessary.

sheffieldnick, 2017-04-24:
Is there any code documentation? I might be able to contribute something simple. e.g., outputting G2/G3 for arcs if there are no matrix transforms at all.

jcoffland, 2017-04-24:
I looked at this again. It's more complicated than I originally thought.  All the moves are represented by straight line segments internally.  This makes for simpler simulation code.  CAMotics turns arc moves in to short line segments here: https://github.com/CauldronDevelopmentLLC/CAMotics/blob/master/src/camotics/machine/MachineLinearizer.cpp#L27  This is fairly early in the process.  The rest of the code deals only with straight line segments in the form of a ``ToolPath`` containing ``Move``s.  ``Move`` would need to be extended to also represent arcs.  A lot of code would need to be updated to handle arc moves directly.

sheffieldnick, 2017-04-24:
Shame! :( Well, thanks for looking into it.

jcoffland, 2017-04-24:
No problem.