CauldronDevelopmentLLC/CAMotics #223: Missing documentation on dplang.org site

sheffieldnick, 2017-04-01:
The documentation for using TPL on [tplang.org](http://tplang.org/#matrix) is very out of date. e.g., no details on how to use matrix functions, even though Github has useful docs written back in 2014: [ CAMotics/doc/tpl_matrices/tplang_matrices.md
](https://github.com/CauldronDevelopmentLLC/CAMotics/blob/master/doc/tpl_matrices/tplang_matrices.md)

jcoffland, 2017-04-01:
When you say the TPL documentation is out of date do you mean anything other than the matrix functions?

sheffieldnick, 2017-04-01:
I don't know - I only found the tplang_matrices.md file on Github by accident. But certainly there is a lot of detail missing on tplang.org:

1. No docs on any of the console.*() functions
2. No docs on Module require()
3. No docs on CAM offset()
4. No docs on DXF open()
5. No mention in Debugging if you can use printf-style. e.g., print("the number: %d\n", number);
6. Tool Movement doesn't describe what 'incremental' actually means? Presumably relative vs. absolute coordinate changes? Two examples would be useful.
7. Some examples would be helpful for using arc(), ideally with graphics.
8. If arc() takes offsets from the current position, shouldn't it be called iarc() for consistency with icut(), irapid() ?

jcoffland, 2017-04-01:
Very helpful input.

Re: 8, you are probably right about calling it ``iarc()`` the reason I called it ``arc()`` is because the "normal" arcs in GCode (i.e. G2/G3) use incremental offsets by default whereas default linear movements (G0/G1) use absolute (or coordinate system relative) movements.  ``arc()`` makes more sense as with relative ``IJK`` offsets, IMO.

``icut()`` is really just ``cut(incremental=true)``.  ``cut()`` is itself ``cut(incremental=false)`` by default, whereas ``arc()`` is ``arc(incremental=true)`` by default.  GCode has the notion of modes where you can be in incremental or incremental arc modes.  I think global modes are confusing and error prone especially in complex programs.